isaac-sim/IsaacLab #1673: [Question] Pytorch Profile Error

Dingjifeng, 2025-01-14:
### Pytorch Profile Error:

Thanks for your work. But when I want to profile with pytorch I encounterd an error:
```
Traceback (most recent call last):
  File "/ssd/public/dingjifeng/IsaacLab/test/lab/tiled_camera.py", line 363, in <module>
    main()
  File "/ssd/public/dingjifeng/IsaacLab/test/lab/tiled_camera.py", line 358, in main
    run_simulator(sim, scene)
  File "/ssd/public/dingjifeng/IsaacLab/test/lab/tiled_camera.py", line 183, in run_simulator
    with torch.profiler.profile(
  File "/home/public/data/dingjifeng/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/profiler/profiler.py", line 706, in __exit__
    self.stop()
  File "/home/public/data/dingjifeng/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/profiler/profiler.py", line 722, in stop
    self._transit_action(self.current_action, None)
  File "/home/public/data/dingjifeng/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/profiler/profiler.py", line 751, in _transit_action
    action()
  File "/home/public/data/dingjifeng/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/profiler/profiler.py", line 206, in stop_trace
    self.profiler.__exit__(None, None, None)
  File "/home/public/data/dingjifeng/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/autograd/profiler.py", line 352, in __exit__
    self.kineto_results = _disable_profiler()
RuntimeError: !stack.empty() INTERNAL ASSERT FAILED at "../torch/csrc/autograd/profiler_python.cpp":969, please report a bug to PyTorch. Python replay stack is empty.
```

And the python code as follows:
```
import argparse
import nvtx
import torch

from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="This script demonstrates the different camera sensor implementations.")
parser.add_argument("--num_envs", type=int, default=4, help="Number of environments to spawn.")
parser.add_argument("--disable_fabric", action="store_true", help="Disable Fabric API and use USD instead.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()
num_envs = args_cli.num_envs
print(f"[INFO]: num_envs = {num_envs}")
# launch omniverse app
app_launcher = AppLauncher({"headless":True,"num_envs":num_envs,"enable_cameras":True})
simulation_app = app_launcher.app

"""Rest everything follows."""

import matplotlib.pyplot as plt
import numpy as np
import os
import torch

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.assets import ArticulationCfg, AssetBaseCfg
from omni.isaac.lab.scene import InteractiveScene, InteractiveSceneCfg
from omni.isaac.lab.sensors import CameraCfg, RayCasterCameraCfg, TiledCameraCfg
from omni.isaac.lab.sensors.ray_caster import patterns
from omni.isaac.lab.terrains import TerrainImporterCfg
from omni.isaac.lab.utils import configclass

##
# Pre-defined configs
##
from omni.isaac.lab.terrains.config.rough import ROUGH_TERRAINS_CFG  # isort:skip
from omni.isaac.lab_assets.anymal import ANYMAL_C_CFG  # isort: skip

@configclass
class SensorsSceneCfg(InteractiveSceneCfg):
    """Design the scene with sensors on the robot."""
[... 241 lines cut ...]
    """Main function."""
    # Initialize the simulation context
    sim_cfg = sim_utils.SimulationCfg(dt=0.005, device=args_cli.device, use_fabric=not args_cli.disable_fabric)
    print(sim_cfg)
    sim = sim_utils.SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view(eye=[0, 3.5, 3.5], target=[1.0, 1.0, 1.0])
    # design scene
    scene_cfg = SensorsSceneCfg(num_envs=args_cli.num_envs, env_spacing=2.0)
    print(scene_cfg)
    scene = InteractiveScene(scene_cfg)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene)

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()
```
Plz Help me if you encountered the same error. Thanks

RandomOakForest, 2025-01-17:
Thanks for submitting this. It doesn't seem `spawn=None` would be allowed in the tiled camera configuration. Could you try following the [example here](https://isaac-sim.github.io/IsaacLab/main/source/overview/sensors/camera.html#tiled-rendering) as a first step?

ykk624, 2025-01-22:
I encountered a same error "self.kineto_results = _disable_profiler()" when I run other model with other project.
Set "with_stack=False" is OK.
我跑其他模型时，也遇到这个错误。将 profiler采集参数里的 with_stack 置为False，即可正常运行。
但是采集到的流水文件，只有算子aten::，没有python侧的代码调用。

如果您解决了这个问题，也可以分享下。

closed 2025-02-15 by RandomOakForest