isaac-sim/IsaacLab #1678: [Proposal] Add an option to "set_external_force_and_torque" to choose where the forces should be applied.

AntoineRichard, 2025-01-15:
### Proposal

Changing the "set_external_force_and_torque" method to allow applying wrenches at a position that's not the center of the articulation.

This could look like so:

```python
    def set_external_force_and_torque(
        self,
        forces: torch.Tensor,
        torques: torch.Tensor,
        positions: torch.Tensor | None = None,
        body_ids: Sequence[int] | slice | None = None,
        env_ids: Sequence[int] | None = None,
    ):
```

When no positions are provided, the implementation uses the current behavior. However, if positions are provided, then it passes them to the PhysX tensor API:

```python
self.root_physx_view.apply_forces_and_torques_at_position(
    force_data=self._external_force_b.view(-1, 3),
    torque_data=self._external_torque_b.view(-1, 3),
    position_data=self._external_wrench_positions_b.view(-1, 3),
    indices=self._ALL_INDICES,
    is_global=False,
)
```

### Motivation
It would be great to be able to apply forces away from the center of the object. This is useful for thruster/propeller based systems such as drones, planes, boats, and satellites. In these systems, the position where the forces are applied can vary depending on the level of realism on wants to have. Also, this can be used to easily implement morphable geometries for drones, satellites, and boats, as there is no need to have different physical configurations: just to apply forces at different positions.

### Alternatives

The current alternative is to create rigid bodies where you want to apply the force, but if you want to change their positions, you need to attach it to a set of joints.

### Checklist

- [x ] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

RandomOakForest, 2025-01-17:
Thanks for posting this proposal. The team will review.

HugoXANA, 2025-02-17:
## Subject: Questions about the apply_forces_and_torques_at_position function

Dear Development Team,
I hope this message finds you well. I am currently exploring the capabilities of the ‘apply_forces_and_torques_at_position’ function in the Omni Physics API and have a few questions to better understand its behavior and best practices for usage.
### Applying Forces at Multiple Positions
If I apply external forces to a rigid body at two different positions, can the simulation environment handle this scenario to simulate the effect of a couple (two equal and opposite forces acting at different points)? I am interested in understanding whether the engine can correctly compute the resulting torque and motion when forces are applied at multiple positions on the same rigid body.
### Handling Multiple External Force Sources
When multiple external force sources are acting on the same rigid body, what is the recommended approach? Should I apply each force individually by calling apply_forces_and_torques_at_position multiple times, or should I calculate the resultant force and torque and apply them in a single call? I would like to know which method is more efficient and accurate in terms of simulation performance and outcome.
### Function Execution in the Simulation
Does the apply_forces_and_torques_at_position function directly affect the simulation environment, or does it merely write to a cache that is then processed by SimulationContext.step? In other words, is the function's effect immediate, or does it require a simulation step to take effect?
I would greatly appreciate any insights you can provide on these questions. Understanding these aspects will help me effectively utilize the function in my simulations.

Thank you for your time and consideration.
Best regards

closed 2025-07-23 by kellyguo11