isaac-sim/IsaacLab #1685: [Bug Report] Terrain Curriculum Features Not Working Properly

H-Hisamichi, 2025-01-16:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Two days ago, I noticed that the random rough terrain curriculum is not functioning correctly. To investigate, I reinstalled IsaacLab and ran some tests. Figures 1–2 show the results of running the `procedural_terrain.py` demo code with the curriculum enabled. Each terrain range is set to 1.0m. It appears that only the random rough terrain curriculum is not working as expected. I have attached the code used for these tests.
Fig. 1:
[image]

Fig. 2:
[image]

Additionally, please refer to Videos 1–4. The robot is spawned with a 90° rotation, but for boxes rough terrain, it falls due to improper interactions with the ground. Video 1 shows flat terrain, Video 2 shows boxes rough with range 0, Video 3 shows boxes rough with range 0.1, and Video 4 shows random rough terrain.
Videos 1:

https://github.com/user-attachments/assets/29fc6383-96d2-47eb-8969-4ec0770eb114

Videos 2:

https://github.com/user-attachments/assets/f417b618-69eb-47bb-98f6-6976804f5113

Videos 3:

https://github.com/user-attachments/assets/501ec9f9-a0de-4b5a-8321-a48f60ab52f9

Videos 4:

https://github.com/user-attachments/assets/c487c86d-7791-4688-a7dc-bb3163c0d5c2

Are these issues caused by a bug?
Thank you for your support.

### Steps to reproduce
procedural_terrain.py:
```
# Copyright (c) 2022-2025, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""
This script demonstrates procedural terrains with flat patches.

Example usage:

.. code-block:: bash

    # Generate terrain with height color scheme
    ./isaaclab.sh -p source/standalone/demos/procedural_terrain.py --color_scheme height

    # Generate terrain with random color scheme
    ./isaaclab.sh -p source/standalone/demos/procedural_terrain.py --color_scheme random

    # Generate terrain with no color scheme
    ./isaaclab.sh -p source/standalone/demos/procedural_terrain.py --color_scheme none

    # Generate terrain with curriculum
    ./isaaclab.sh -p source/standalone/demos/procedural_terrain.py --use_curriculum

    # Generate terrain with curriculum along with flat patches
    ./isaaclab.sh -p source/standalone/demos/procedural_terrain.py --use_curriculum --show_flat_patches

"""

"""Launch Isaac Sim Simulator first."""

import argparse

from omni.isaac.lab.app import AppLauncher

# add argparse arguments
[... 351 lines cut ...]
        "wave_terrain": terrain_gen.HfWaveTerrainCfg(
            amplitude_range=(0.0, 0.2), num_waves=40.0 #border_width=0.25
        ),
    },
)
"""Rough terrains configuration."""
```

-->

### System Info

Describe the characteristic of your environment:

- Commit:
- Isaac Sim Version: [4.2.0]
- OS: [Ubunt 22.04.5 LTS]
- GPU: [RTX A4000]
- CUDA: [12.2]
- GPU Driver: [535.183.01]

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-01-17:
Thank you for posting this. I see multiple questions here and may have to move this into a discussion. I'll review with the team and aim to follow up soon.

H-Hisamichi, 2025-01-18:
Dear @RandomOakForest,

Thank you for starting the investigation. As additional information, the default USD pose of my robot is shown in the image. After spawning in Isaac Lab, I rotate the entire robot by 90° and reset the joint angles to form a cylindrical shape.

USD pose:

[image]