isaac-sim/IsaacLab #1687: [Proposal] collision groups for collision detection

junwang048bot, 2025-01-16:
### Proposal

It would be great if we could define the collision group for collision detection like all the other simulators in isaaclab.

### Motivation

I am currently frustrated on solving the self-collision. Once I have enabled the self-collision, the behaviors for the robots would change and become so weird they even don't have a way to filter the collision between the connected links.

### Alternatives

A clear and concise description of any alternative solutions or features you've considered, if any.

### Additional context

Add any other context or screenshots about the feature request here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

- [x] Criteria 1 able to filter given collision group
- [x] Criteria 2 would not have a behavior change when enable the self-collision

KyleM73, 2025-01-16:
What robot model are you using? Are the collision meshes intersecting by default? Oftentimes if the robot has intersecting collision meshes I will get weird behavior until I fix the model so there isn't. Collision filtering doesn't stop Isaac Sim from detecting the contacts to my knowledge, it just doesn't incorporate them in the physics step, so it is more effective to iterate on the model first. If your model is a URDF, a quick viewer like pybullet or this browser-based one may be useful to visualize the contact meshes: https://gkjohnson.github.io/urdf-loaders/javascript/example/bundle/index.html

junwang048bot, 2025-01-16:
if you don't enable the self-collision for the robot, all the contacts between the links of the robot would not be calculated inside physx which would fix the issue. Even if the collision meshes do not intersect by default, the collision type of the mesh would affect whether you are using a convex-hull / convex-decomposition. And the self-collision enable don't have the same behaviors in isaaclab and isaacsim which seems weird too.

RandomOakForest, 2025-01-18:
Thank you for posting this proposal. The team will review it.