isaac-sim/IsaacLab #1688: [Question] Missing ArticulationCfg Attribute

htchr, 2025-01-16:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

Hello,

I'm trying to work through the demonstration from a [recent Isaac Lab Office Hours](https://www.youtube.com/watch?v=_L5c3LwZhL0).

I've converted a custom URDF to USD using the Isaac Sim URDF importer.

I've drafted a basic ArticulationCfg:

```python
NOODLEBOT_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path="/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/noddlebot/noodlebot_simple.usd",
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            rigid_body_enabled=True,
            max_linear_velocity=1000.0,
            max_angular_velocity=1000.0,
            max_depenetration_velocity=100.0,
            enable_gyroscopic_forces=True,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=False,
            solver_position_iteration_count=4,
            solver_velocity_iteration_count=0,
            sleep_threshold=0.005,
            stabilization_threshold=0.001,
        ),
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.06), 
        joint_pos={".*": 0.0}
    ),
    actuators={
        "position_joints": ImplicitActuatorCfg(
            joint_names_expr=[".*link.*"],
            effort_limit=400.0,
            velocity_limit=1.0,
            stiffness=1000.0,
            damping=0.0,
        ),
        "velocity_joints": ImplicitActuatorCfg(
            joint_names_expr=[".*frame.*"],
            effort_limit=40000.0,
            velocity_limit=9.0,
            stiffness=0.0,
            damping=100000.0,
        ),
    },
)
```

I've created the most simple DirectRLEnvCfg according to the YouTube tutorial:

```python
@configclass
class NoodlebotEnvCfg(DirectRLEnvCfg):
    # env
    decimation = 2              # number of time steps between actions
    episode_length_s = 5.0      # episode time max
    action_space = 0            # number of actions the neural net returns
    observation_space = 0       # number of observations fed into the neural net
    state_space = 0             # observations to be used in actor-critic training

    # simulation
    sim: SimulationCfg = SimulationCfg(dt=1/60, render_interval=decimation)

    # robot
    robot_cfg: ArticulationCfg = NOODLEBOT_CFG.replace(prim_path="/World/envs/env_.*/Robot")
    position_joint_names = [".*link.*"]
    velocity_joint_names = [".*frame.*"]

    # scene
    scene: InteractiveSceneCfg = InteractiveSceneCfg(num_envs=4096, env_spacing=4.0, replicate_physics=True)
```

When I attempt to run with `python3 source/standalone/environments/random_agent.py --task Noodlebot_1-Direct-v0 --num_envs 16` I receive this error:

```bash
Traceback (most recent call last):
  File "/workspace/isaaclab/source/standalone/environments/random_agent.py", line 68, in <module>
    main()
  File "/workspace/isaaclab/source/standalone/environments/random_agent.py", line 46, in main
    env = gym.make(args_cli.task, cfg=env_cfg)
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/noodlebot/noodlebot_directenv.py", line 42, in __init__
    super().__init__(cfg, render_mode, **kwargs)
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py", line 160, in __init__
    self._window = self.cfg.ui_window_class_type(self, window_name="IsaacLab")
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/ui/base_env_window.py", line 87, in __init__
    self._build_debug_vis_frame()
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/ui/base_env_window.py", line 241, in _build_debug_vis_frame
    self._create_debug_vis_ui_element(name, elem)
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/ui/base_env_window.py", line 402, in _create_debug_vis_ui_element
    if elem.has_debug_vis_implementation
AttributeError: 'ArticulationCfg' object has no attribute 'has_debug_vis_implementation'
```

I haven't been able to find any information about this error so I would really appreciate any help making sense of it so I can start training my robot.

Thank you

htchr, 2025-01-17:
I found the answer.

In the `DirectRLEnv` class, the `Articulation` class had been autocorrected to `ArticulationCfg`.
Using `Articulation` fixed the issue:

```python
class NoodlebotEnv(DirectRLEnv):
    cfg: NoodlebotEnvCfg

    def __init__(self, cfg: NoodlebotEnvCfg, render_mode: str | None = None, **kwargs):
        super().__init__(cfg, render_mode, **kwargs)
        self._position_joint_idx, _ = self.Noodlebot.find_joints(self.cfg.position_joint_names)
        self._velocity_joint_idx, _ = self.Noodlebot.find_joints(self.cfg.velocity_joint_names)
        print("[DEBUG] init complete")

    def _setup_scene(self):
        print("[DEBUG] setup scene starting")
        self.Noodlebot = Articulation(self.cfg.robot_cfg)
        ...
```

closed 2025-01-17 by htchr

linchangyi1, 2025-02-06:
Hi @htchr, I also encountered this issue with a customized USD from URDF. I just checked and found that "Articulation" was used in the environment. Could you please explain how you fixed that with more detail~

htchr, 2025-02-06:
> Hi [@htchr](https://github.com/htchr), I also encountered this issue with a customized USD from URDF. I just checked and found that "Articulation" was used in the environment. Could you please explain how you fixed that with more detail~

Switching from `ArticulationCfg` to `Articulation` completely fixed my issue.

zxfever, 2025-02-25:
> Hi [@htchr](https://github.com/htchr), I also encountered this issue with a customized USD from URDF. I just checked and found that "Articulation" was used in the environment. Could you please explain how you fixed that with more detail~

hey bro, face the same error, and also have checked that the `Articulation` used in the environment, did you fix the problem?