isaac-sim/IsaacLab #1689: [Question] error importing opencv after the simulation app is launched

SantiDiazC, 2025-01-17:
Hi, 

I created a DirectRL environment and was debugging some of its functions. I wanted to test the images obtained as observations from the camera sensor using OpenCV's imshow function. It works fine if you import the OpenCV library before launching the simulation app. However, if you import it afterward, the following error occurs:

```error
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```
I don't know if it is a bug or if it is an expected behavior.

Thank you

RandomOakForest, 2025-01-18:
Thank you for posting this question. This is typically an issue with the setup of your environment (opencv, qt libraries, etc.). If you can share more detailed steps to reproduce this issue we could provide further guidance. How did you installed Isaac Lab and Isaac Sim?

SantiDiazC, 2025-01-21:
Hi,
Thank you for you reply! I installed isaac lab from binaries following the installition guide in the documentation. I have isaac sim 4.2.0 and Isaac lab 1.4.0.
I was testing my environment using this simple script:

```python
import argparse
import cv2
from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on running the cartpole RL environment.")
parser.add_argument("--num_envs", type=int, default=16, help="Number of environments to spawn.")

# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import torch
import numpy as np

from omni.isaac.lab.envs import ManagerBasedRLEnv

from omni.isaac.lab_tasks.direct.my_env. my_env import MyBaseEnv, MyBaseEnvCfg

def main():
    """Main function."""
    # create environment configuration
    env_cfg = MyBaseEnvCfg()
    env_cfg.scene.num_envs = args_cli.num_envs
    # setup RL environment
    env = MyBaseEnv(cfg=env_cfg)

    # simulate physics
    count = 0
    while simulation_app.is_running():
        with torch.inference_mode():
            # reset
            if count % 300 == 0:
                count = 0
                env.reset()
                print("-" * 80)
                print("[INFO]: Resetting environment...")
            # sample random actions
            joint_efforts = torch.randn_like(env.robot_dof_targets)
            # step the environment
            obs, rew, terminated, truncated, info = env.step(joint_efforts)
            # check the camera output
            if count % 100 == 0 and count > 0:
                img = obs["policy"][0].cpu().numpy().transpose(0, 2 , 3, 1)
                rgb_img = cv2.cvtColor(img[0], cv2.COLOR_BGR2RGB)
                cv2.imshow("rgb image", rgb_image)
                cv2.waitKey(0)
                cv2.destroyAllWindows()
            # update counter
            count += 1

    # close the environment
    env.close()

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()
```

The observations in my environment class are the images from a camera sensor (not tiled camera):

```python
def _get_observations(self) -> dict:
        self.compute_intermediate()
        if self.cfg.use_visual_obs:
            img_f = self.camera.data.output['rgb'] #(num_envs, H, W, 4)
            # img = img[:,:,:,:-1]
            img_f = img_f[:,:,:,:]
            img_f = torch.permute(img_f,(0, 3, 1, 2))
            
            observations = {"policy": img_f}
         return observations
```
The error occurs when I import the library after the simulation app is launched:
before launching (Works fine):
```python
import argparse
import cv2
from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on running the cartpole RL environment.")
parser.add_argument("--num_envs", type=int, default=16, help="Number of environments to spawn.")

# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app
```

After launching (Error occurs):
```python
import argparse
from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on running the cartpole RL environment.")
parser.add_argument("--num_envs", type=int, default=16, help="Number of environments to spawn.")

# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import torch
import numpy as np
import cv2
.
.
.
```
So, I was curios about this behavior, like you have to import always first opencv or the import order should not affect it and therefore this is a bug.
Thank you

RandomOakForest, 2025-03-07:
Following up on this. Do you still have this issue with the latest versions of Isaac Lab and Isaac Sim? Also, be sure to use the recommended driver. If the issue persist, please file a _bug report_. Thank you.

closed 2025-03-07 by RandomOakForest