isaac-sim/IsaacLab #1690: [Bug Report] Contact penetration

elle-miller, 2025-01-17:
I am simulating a franka arm with a custom-made object. I am using convex decomposition for the object's colliders, which is visualised in the video. 

I would like to improve the contact collision behaviour, many times throughout the video **the gripper fingers penetrate the sides of the object**, and it becomes "stuck" there.

In addition, I am using custom contact sensors on the gripper fingers. When I try and filter the contacts to only measure the object's forces, I receive the following message `GPU Contact filter for collider is not supported for object ....`. I had this working fine for default objects, so seems it is related to the convex decomposition.

Let me know what options I have for improving this behaviour. Thank you! :)

https://github.com/user-attachments/assets/b68b02e9-8efb-4b4f-9aac-8f5b9d5025e5

### System Info

Describe the characteristic of your environment:

- Commit: Isaac Lab 1.2.0
- Isaac Sim Version: 4.2.0 
- OS: Ubuntu 20.04
- GPU: RTX 2080
- CUDA: 12.2
- GPU Driver: 535.183.01

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Have the robot pick up the object without visibly penetrating the sides

RandomOakForest, 2025-01-18:
Thank you for posting this. It may be a limitation in the PhysX extension, we are reviewing this issue.

elle-miller, 2025-01-20:
Thank you @RandomOakForest, looking forward to your comments

RandomOakForest, 2025-01-24:
After discussing with the team, here are a few suggestions. We think the sides of the cube-like object are a bit thin, maybe it would help making them thicker. Also, convex decomposition may not be as good as an SDF mesh in this case. Finally, try decreasing the simulation _dt_ (time between steps) while increasing the position and velocity iterations. 

Thanks for your interest in Isaac Lab. Regarding this post, we think is not precisely something we identify as a bug just yet, so I will move it to our _Discussions_ section.