isaac-sim/IsaacLab #1691: [Question] How can I control the output log level using AppLauncher?

chrisnota, 2025-01-17:
### Question

IsaacLab and IsaacSim are very "noisy" by default, in that they print a lot to the command line. In some cases, I want to suppress this. I'm instantiating a custom environment using the following logic:

```python
    def _make_env(self) -> gym.Env:
        # Isaac Lab supports rgb_array render mode for recording videos.
        # This requires the cameras to be enabled.
        enable_cameras = self.gym_make_kwargs.get("render_mode", None) == "rgb_array"

        # Create the AppLauncher. This import is heavy-weight,
        # so we import here instead of at the top of the file.
        from omni.isaac.lab.app import AppLauncher

        AppLauncher(headless=self.headless, enable_cameras=enable_cameras)

        # Make sure the custom environments are registered.
        # This will not work before the AppLauncher is created.
        if not self.registered:
            register_envs()
            self.registered = True

        # Create environment configuration.
        # This will not work before the AppLauncher is created.
        from omni.isaac.lab_tasks.utils import parse_env_cfg

        env_cfg = parse_env_cfg(self.task, device=self.device, num_envs=self.num_envs)

        # create environment
        return gym.make(
            self.task,
            cfg=env_cfg,
            **self.gym_make_kwargs,
        )
```

How can I control the log level? I would like to set it Error, or even disable it altogether.

RandomOakForest, 2025-01-18:
Thanks for submitting this post. There may be a better way to do this in the upcoming Isaac Sim 4.5 release, though there are some guidelines in [this post in the Isaac Sim forums](https://forums.developer.nvidia.com/t/isaac-sim-log-level-to-many-log-messages/308545).

chrisnota, 2025-01-21:
Thanks - I'll give this a try!

edubya, 2025-03-04:
Since upgrading to 4.5 I started seeing a lot more debug messages when the app is launched. I cannot figure out how to suppress these debug messages or any following INFO messages.

I also noticed that I get debug messages from matplotlib after app launch, so something is going in and setting the log level globally to debug and I cannot figure out where.

I am running headless with cameras enabled, so I tried modifying the experience files as suggested in https://forums.developer.nvidia.com/t/isaac-sim-log-level-to-many-log-messages/308545.

This did not work so I also tried modifying the kit_args passed to AppLauncher, as suggested in the same post. I can see _kit_args are updated but this did not do anything ... ? I also tried to disable logging entirely using omni.log as suggested in a brief answer to another post here.

Here is what I did to try setting kit_args:
```
    import argparse
    opts = argparse.Namespace()
    opts.kit_args='--/log/level=error --/log/fileLogLevel=error --/log/outputStreamLevel=error'
    opts.headless=True
    opts.enable_cameras=True
    opts.device='cuda:0'
    
    launcher = AppLauncher(opts)
```

Tried to change logging in isaaclab.python.headless.rendering.kit (as well as python.headless.kit and python.kit):
```
[settings]
# Note: This path was adapted to be respective to the kit-exe file location
app.versionFile = "${exe-path}/VERSION"
app.folder = "${exe-path}/"
app.name = "Isaac-Sim"
app.version = "4.5.0"

# Disable print outs on extension startup information
# this only disables the app print_and_log function
app.enableStdoutOutput = false

log.level = "error"
log.fileLogLevel = "error"
log.outputStreamLevel = "error"

# set the default ros bridge to disable on startup
isaac.startup.ros_bridge_extension = ""
```

edubya, 2025-03-04:
Running this before launching app helps a lot:
```
import logging
logger = logging.getLogger()
logger.setLevel(logging.ERROR)
for hndlr in logger.handlers:
    logger.removeHandler(hndlr)
```

RandomOakForest, 2025-03-07:
Thanks for following up. If this issue persists in the latest versions of Isaac Lab and Isaac Sim, please open a new question or a bug report with details on how to reproduce.

closed 2025-03-07 by RandomOakForest

bikcrum, 2025-08-31:
In IsaacLab, you can also use `--kits_args "--/log/level=error --/log/outputStreamLevel=error --/log/fileLogLevel=error` to suppress the warning and only show errors.

Ref: https://github.com/isaac-sim/IsaacLab/issues/937#issuecomment-3240461598