isaac-sim/IsaacLab #1702: [Bug Report] Locked pos or axis only applies to first environment

elle-miller, 2025-01-20:
I wish to lock the rotation axis of my object. It works for 1 environment when I test it on the viewer (see picture). 

[image]

However, when I scale up the environments it is not being applied to other objects (see video). You can also see that the collisions disappear for this object for some reason. I am using `replicate_physics=True`.

Any help would be greatly appreciated!

https://github.com/user-attachments/assets/00164d21-478a-450c-bb50-51ea4967b0a5

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-01-21:
Thank you for posting this. The team will review. We should be able to follow up soon.

RandomOakForest, 2025-01-23:
After discussing with the team, this may be something we have to repro. Have you tried using `replicate_physics=False` and see the same effect? Thanks.

RandomOakForest, 2025-04-18:
Following up, I'll close the issue for now. If you still face this issue, please open a new [discussion](https://github.com/isaac-sim/IsaacLab/discussions). Thank you.

closed 2025-04-18 by RandomOakForest