isaac-sim/IsaacLab #1708: [Question] How to control robot

MrVBian, 2025-01-22:
I asked the same question on the forum：[https://forums.developer.nvidia.com/t/physx-error-pxarticulationjointreducedcoordinate-setdrivetarget/320512](https://forums.developer.nvidia.com/t/physx-error-pxarticulationjointreducedcoordinate-setdrivetarget/320512)

### Question

After starting the sim based on isaac lab, I need to accept the joint value of the ros topic and use it as the target to control the simulated robot arm.
Therefore, I try to do the following:
```python
    while simulation_app.is_running():
        sim.step(render=True)
		# Omit the steps to get the joint values in the ros topic.
		# The result is stored in the targetPositions[]
        action = ArticulationAction(joint_positions=np.array(targetPositions))
        articulation.apply_action(action)
```
This will actually report an error. I don't know what to do to achieve my business goals.
What can I do to fix my problem, I can't find how to turn off PxSceneFlag::eENABLE_DIRECT_GPU_API. 
Thank you for your reply!

### Steps to Reproduce

https://docs.omniverse.nvidia.com/isaacsim/latest/how_to_guides/robots_simulation.html#

Referring to the position-control code in the link above, I was able to run it normally in the script editor of isaac sim4.2. When I tried to embed this code block in a standalone script in the isaac lab environment, an error was reported at the dc.set_articulation_dof_position_targets stage

```python
from omni.isaac.dynamic_control import _dynamic_control
import numpy as np
dc = _dynamic_control.acquire_dynamic_control_interface()
articulation = dc.get_articulation("/Franka")
# Call this each frame of simulation step if the state of the articulation is changing.
dc.wake_up_articulation(articulation)
joint_angles = [np.random.rand(9) * 2 - 1]
dc.set_articulation_dof_position_targets(articulation, joint_angles)
```

### Error Messages

```error
PhysX error: PxArticulationJointReducedCoordinate::setDriveTarget(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, FILE /builds/omniverse/physics/physx/source/physx/src/NpArticulationJointReducedCoordinate.cpp, LINE 304
```

Mayankm96, 2025-01-22:
Please check our tutorials: https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/run_articulation.html#stepping-the-simulation

MrVBian, 2025-01-22:
> Please check our tutorials: https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/run_articulation.html#stepping-the-simulation

Thank you for your reply. I have another question I would like guidance on.
Instantiating Robot in the tutorial:
```python
    # Articulation
    cartpole_cfg = CARTPOLE_CFG.copy()
    cartpole_cfg.prim_path = "/World/Origin.*/Robot"
    cartpole = Articulation(cfg=cartpole_cfg)
` ` `
How to instantiate the attributes in a scene into Articulation by prim_path. So I can use the robot properties that are already configured in the UI.

RandomOakForest, 2025-03-07:
@MrVBian if you still have another question, please open a new issue. Thanks.

closed 2025-03-07 by RandomOakForest

forward1023, 2025-12-28:
你好，请问你的问题PhysX error: PxArticulationJointReducedCoordinate::setDriveTarget(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, FILE /builds/omniverse/physics/physx/source/physx/src/NpArticulationJointReducedCoordinate.cpp, LINE 304解决了吗，怎么解决的？