isaac-sim/IsaacLab #1709: [Bug Report] Issue with static type checking in VSCode

Mayankm96, 2025-01-22:
### Describe the bug

After updating my VSCode, I am unable to automatically jump to the right IsaacSim/IsaacLab code from the fork of my extension template. On inspection of the VSCode `settings.json` file, there is a warning there saying:

```
'python.analysis.extraPaths' cannot be set when a pyrightconfig.json or pyproject.toml is being used.settings.jsonsettingsNotOverridable
```

Since IsaacLab project uses a `pyproject.toml` in the top, the above issue makes sense and is also documented [here](https://github.com/microsoft/pylance-release/wiki/Settings.json-overridden-by-Pyrightconfig.json-or-Pyproject.toml).

I managed to solve the issue by following the steps [here](https://micropython-stubs.readthedocs.io/en/main/22_vscode.html). It mainly says to assign the analysis paths to the variable `extraPaths` under `[tools.pyright]` in the `pyproject.toml` file.

### Steps to reproduce

Update your VSCode and you should see the same issue. My PC has multiple versions of IsaacLab installed so it always end up pointing to the wrong one.

### System Info

- Commit: Release 1.2 (but pretty sure it exists in other ones)
- Isaac Sim Version: 4.2
- OS: Ubuntu 20.04
- GPU: RTX 4090

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] VSCode static checking should work properly

kellyguo11, 2025-07-07:
@Toni-SM do you know if this is still an issue with the current VSCode setup?

maxkra15, 2026-09-22:
Thanks @Mayankm96 — this was fixed in [#6644](https://github.com/isaac-sim/IsaacLab/pull/6644). Please reopen if VS Code still resolves the wrong checkout after running `uv run isaaclab --editor` on current Isaac Lab.

closed 2026-09-22 by maxkra15