isaac-sim/IsaacLab #1718: [Question] Camera and tiled camera have different type for data.info

SantiDiazC, 2025-01-24:
Hi,

I was testing some environment with two different cameras using instance segmentation as observation,
and I noticed that for the two sensors the data.info variable is a list and a dictionary for the camera and the tiled camera respectively.
with the data.info I was intending to get the labels of the segmentation mask but this difference makes you need to modify the code depending of the sensor used which I think is inefficient.

so my question is: is there any plan to unify this feature? it will be great to have same data types to quickly implement and debug the code with different sensors. or is any reason to have different data types at the moment?

Thank you in advanced!

kellyguo11, 2025-01-26:
Thanks for bringing this to our attention. I agree we should unify the data types for the two classes. We'll look into this!

AntoineRichard, 2026-08-27:
@SantiDiazC good catch on the inconsistent `data.info` type between the two camera sensors — that mismatch was real.

The 3.0 renderer refactor unified the stack: `TiledCamera` now subclasses `Camera` and both share a single `CameraData` whose `info` field is `dict[str, Any] | None`, keyed by output name and populated per data type by each renderer backend (`source/isaaclab/isaaclab/sensors/camera/camera_data.py`). No list-typed `info` remains in the tree, so semantic and instance label maps are accessed identically for both sensors.

Closing this out — happy to reopen if you hit a type mismatch on 3.x.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard