isaac-sim/IsaacLab #1720: [Proposal] Extend `ContactSensorData` by `force_matrix_w_history` attribute

lukasfro, 2025-01-24:
### Proposal

Similar to the `net_forces_w_history` attribute that keeps a history of `net_forces_w`, I propose to add a `force_matrix_w_history` attribute storing the history of `force_matrix_w` when using the `ContactSensor` with filtered bodies.

### Motivation

Depending on the decimation factor, we noticed that it is critical to consider the full history of the contact sensor. This is in accordance with how the `illegal_contact` termination is computed [here](https://github.com/isaac-sim/IsaacLab/blob/21173c3e933a9236d6eb5eb2adb903494ba9a418/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/terminations.py#L154). When filtering the collision bodies, this functionality does not exist currently. I'm happy to open a PR for this feature. We have been using this internally already.

### Alternatives

There are no real alternatives to this.

### Additional context

---

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-01-24:
Thank you for posting this proposal. The team will review.

RandomOakForest, 2025-01-27:
Following up on this feature request. The team would welcome a PR may you have time to submit one. Thanks.

commit 1e273226 (2025-07-16): Extends `ContactSensorData` by `force_matrix_w_history` attribute (#2916)

# Description

This is a follow-up to #1746 by @lukasfro — thanks for the great work
there!

I’ve been using this feature and found it really helpful. Since the only
remaining request was to resolve merge conflicts and the PR has been
quiet for a bit, I went ahead and rebased it on the latest `main` to
help move things forward.

I branched directly off @lukasfro’s [original
branch](https://github.com/lukasfro/IsaacLab/tree/feature/force_matrix_w_history)
to preserve their authorship, and only applied the changes needed  [... cut]
closed 2025-07-16 by kellyguo11