isaac-sim/IsaacLab #1722: [Question] Trying to use RigidPrimView to get position of a prim in Observation manager

vluduk, 2025-01-25:
### Question

Hi there!
I am trying to use RigidPrimView.get_world_pose() to get position of a moving car in Observation manager. It seems like the manager does not support the usage of RigidPrimView, since you can create only ObservationTerms in the obsManager class. RigidPrimView needs to be either initilized every time, or be put in a global variable.

How it can be done? Or how can i get a prim`s actual moving position using observationManager?

StrainFlow, 2025-01-28:
Does the ArticulationData class give you what you need? 

https://isaac-sim.github.io/IsaacLab/main/source/api/lab/omni.isaac.lab.assets.html#omni.isaac.lab.assets.ArticulationData

vluduk, 2025-01-29:
It does, thanks! Although i have a different problem now.
I've initialized a RigidContactView to check for collisions and there's no way i can put it in ObservationManager. Is there any way to access initialized Views from the env? I see it's possible to add some rigid bodies to the scene:
...
elif isinstance(asset_cfg, RigidObjectCfg):
        self._rigid_objects[asset_name] = asset_cfg.class_type(asset_cfg)
elif isinstance(asset_cfg, RigidObjectCollectionCfg):
...
(from IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py)
and they initialize a RigidPrimView but they don't track contact forces? and also create a new sim view??
(self._physics_sim_view = physx.create_simulation_view(self._backend from IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/rigid_object_collection/rigid_object_collection.py, but instead of backend it should be frontend if we believe the source code in extsPhysics\omni.physics.tensors\omni\physics\tensors\impl\api.py?)
is it possible to access a RigidContactView from ObservationManager without any workarounds and without initializing it every iteration?

Mayankm96, 2025-01-30:
Instead of using `RigidContactView`, we recommend using the contact sensor which uses the PhysX contact view class directly and has additional features (such as lazy sensor update, history etc.)

vluduk, 2025-01-31:
It works, thanks!

closed 2025-01-31 by vluduk