isaac-sim/IsaacLab #1725: [Proposal] Measured Torque and Forces (FT Sensors)

SuomiKP31, 2025-01-25:
### Proposal

Implement FT Sensors in Isaac Sim, which will make measured torque and forces on joints a bit easier to get in Isaac Lab DirectRL Envs.

### Motivation

Currently, I'm using fixed PhysX joints to mimic an FT Sensor in MuJoCo, proposed by https://github.com/isaac-sim/IsaacLab/discussions/139. It works pretty well, but I did not find this way before a good amount of research and engineering that I wish I didn't have to do.

Plus, using this API means you have to initialize two simulation views. `omni.isaac.core.articulations.ArticulationView` and `omni.isaac.lab.lab.assets.Articulation` both call `omni.physics.tensors.create_simulation_view(self._backend)` when initialized.

Getting measured torque and forces on feet sensors are pretty important for biped robotics learning tasks. Although the alternative is perfectly usable now, I wish we could have cleaner ways of accessing those information.

### Alternatives

You can use the OG ArticulationView APIs to get measured torque (effort) & forces on joints like this:

```
        self.FT_views = ArticulationView(prim_paths_expr="/World/envs/env_.*/Robot/base_link")
        self.FT_views.initialize(self.robot._physics_sim_view) # LFT / RFT index = 31, 32
        self.LFT_index, self.RFT_index = self.FT_views.get_joint_index("L_Foot_Link"), self.FT_views.get_joint_index("R_Foot_Link")

...
        # Read sensors
        ft_forces = self.FT_views.get_measured_joint_forces() # type: ignore
        self.ft_left_foot = ft_forces[:, self.LFT_index + 1, :3] # Need the plus one for the correct joint index 
        self.torque_left_foot = ft_forces[:, self.LFT_index + 1, 3:] # Annoying and unreadable!
        self.ft_right_foot = ft_forces[:, self.RFT_index + 1, :3]
        self.torque_right_foot = ft_forces[:, self.RFT_index + 1, 3:]
```
For context, `R_Foot_Link` and `L_Foot_Link` are Fixed Joints. They were FT Sensors in MuJoCo, and they got automatically created when I imported my MJCF xml into Isaac Sim, which indicates that this is the default way to implement FT sensors in Isaac.

```
<body name="L_Foot_Link" pos="0 0 0">
    <inertial fullinertia="0.003386409 0.012600125 0.014296 1.1237E-05 -0.000526477 -1.976E-06" mass="2.350" pos="0.016 -0.00008 -0.1206" />
    <geom class="viz" mesh="Foot_Link" rgba="0.85938 0.07813 0.23438 1" />
    <geom class="cls" pos="0.03 0 -0.145" size="0.15 0.085 0.0135" type="box" />
    <geom class="cls" pos="-0.08875 0 0" quat="0.707107 0 -0.707107 0" size="0.045 0.03125" type="cylinder" />
    <site class="FTsensor" name="LF_FT" pos="0 0 -0.09" />
</body>
```

### Additional context

Alternatively, putting a get_measured_forces() API in Isaac Lab's Articulation API will also work (at least for me).

### Checklist

- [X] I have checked that there is no similar issue in the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] FT Sensor Class implemented, 
- [ ] or API similar to get_measured_forces() is implemented in Isaac Lab's Articulation API

kellyguo11, 2025-01-26:
Thanks for the proposal, we are working on adding the force related APIs to Isaac Lab.

SuomiKP31, 2025-02-25:
Adding to the usability of this alternative,
it seems to cause a PhysX error
```
2025-02-25 22:02:23 [26,227ms] [Error] [omni.physx.plugin] PhysX error: PxArticulationLink::setGlobalPose():
it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, 
FILE C:\g\101116487\physx\source\physx\src\NpArticulationLink.cpp, LINE 353
```
which does not impact the usability of the system until more than 96/100 articulation views are created. I'm guessing there are data races between the two PhysX views.

When I try run with `num_env=128`, PhysX stalls with `Failed to create simulation view: no active physics scene found`.

For now 96 works for me, but I'd like to see FT sensors supported in the future!

AntoineRichard, 2026-08-27:
Thanks for the proposal, @SuomiKP31 — it's shipped. Isaac Lab now has a first-class `JointWrenchSensor` / `JointWrenchSensorCfg`, exported from `isaaclab.sensors`, reporting measured joint `force` and `torque` (in the incoming joint frame by default) with no second `ArticulationView` needed. It landed in https://github.com/isaac-sim/IsaacLab/pull/5412, with backend implementations for PhysX (https://github.com/isaac-sim/IsaacLab/pull/5458) and OvPhysX (https://github.com/isaac-sim/IsaacLab/pull/5706).

Closing as implemented — please reopen if the sensor doesn't cover your FT use case.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard