isaac-sim/IsaacLab #1729: Issue with the lighting conditions while using Isaac lab

balaprenithreddy, 2025-01-26:
Hello,

I have an object and I was trying to simulate the lighting conditions which are similar to the space environment (where only few faces of the object should be illuminated). The approach I took was very similar, I just made few modifications to the run_usd_camera.py in standalone\tutorials\04_sensors example file, like spawning the object from the USD file instead of create objects. When I run the script (Isaac Sim 4.2 python), all the faces of my object are illuminated, but when I checked the USD file manually using Isaac Sim, only few faces are illuminated.  I haven't faced this issue in the earlier versions of Isaac Lab. 

Video from Isaac Sim 4.2 python
https://github.com/user-attachments/assets/7e43bf14-9b1b-496c-8d8e-2b6850d1408c

Video from Isaac Sim 4.2 - This is how I want my lighting conditions to be.
https://github.com/user-attachments/assets/37bdb295-3e6a-4275-9c6c-a12960f9ce0e

Thanks

RandomOakForest, 2025-01-27:
Thanks for posting this. It seems the light source is different in the 2 examples. This is mostly an Isaac Sim question. Have you checked [this doc on lighting](https://docs.omniverse.nvidia.com/isaacsim/latest/gui_tutorials/tutorial_intro_environment_setup.html#lighting)? Also, there may be a solution to this in [the Isaac Sim forum](https://forums.developer.nvidia.com/c/omniverse/simulation/69).

balaprenithreddy, 2025-01-27:
In Isaac Sim, its working as expected but when I am trying to run the custom script from Isaac Lab it illuminates the entire object. Whereas in Isaac sim it will only illuminate few faces.

RandomOakForest, 2025-03-07:
Thanks for following up. We may need more information to reproduce this issue. If you have found this issue persists in the latest versions of Isaac Lab and Isaac Sim, with the recommended driver, please open a _bug report_ so we can investigate. Thanks.

closed 2025-03-07 by RandomOakForest