isaac-sim/IsaacLab #1737: [Bug Report] Isaac Lab causes errors when loading robots manually from UI

MuammerBay, 2025-01-27:
### Describe the bug

When launching the script from the CLI "python source\standalone\tutorials\00_sim\create_empty.py" from the Isaac Lab Tutorials and trying to access a robot manually within the simulation, i.e. in my case from the Simulation UI: Create -> Isaac -> Robots -> Wheeled Robots -> AgilexRobotics -> Limo, it causes errors shown below. This holds true for other robots as well ( Create -> Isaac -> Robots ). The robot is getting imported but the simulation stops and can not be resumed.

When running the humanoid example: "python source/standalone/workflows/skrl/train.py --task Isaac-Humanoid-Direct-v0  --num_envs 16" , interestingly I can successfully import the robots from above, the RL training stops for a second, the humanoids start to "fly" and then the training restarts from 0% (as seen in the 2. logs). The simulation itself doesn't stop unlike the "create_empty.py" example.

### Steps to reproduce

1. Scenario
conda activate isaaclab
cd ..\IsaacLab\
python source\standalone\tutorials\00_sim\create_empty.py
Simulation UI: Create -> Isaac -> Robots -> Wheeled Robots -> AgilexRobotics -> Limo

2. Scenario
conda activate isaaclab
cd ..\IsaacLab\
python source/standalone/workflows/skrl/train.py --task Isaac-Humanoid-Direct-v0  --num_envs 16
Simulation UI: Create -> Isaac -> Robots -> Wheeled Robots -> AgilexRobotics -> Limo

```
(isaaclab) C:\IsaacLab\IsaacLab>python source\standalone\tutorials\00_sim\create_empty.py
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: C:\IsaacLab\IsaacLab\source\apps\isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'c:/users/murme/miniconda3/envs/isaaclab/lib/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: c:/users/murme/miniconda3/envs/isaaclab/lib/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20250127_174315.log
2025-01-27 16:43:15 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 566.36        | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3090          | Yes: 0 |     | 24540   MB | 10de      | 48e00000.. |
|     |                                  |        |     |            | 2204      | 923f3a8c.. |
|     |                                  |        |     |            | 2b        |            |
|=============================================================================================|
| OS: Windows 10 Home, Version: 10.0 (22H2), Build: 19045, Kernel: 10.0.19041.5369
| Processor: AMD Ryzen 5 5600X 6-Core Processor              | Cores: 6 | Logical: 12
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 32694 | Free Memory: 20816
| Total Page/Swap (MB): 40886 | Free Page/Swap: 21804
|---------------------------------------------------------------------------------------------|
2025-01-27 16:43:20 [4,805ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
[INFO]: Setup complete...
2025-01-27 16:43:37 [22,255ms] [Error] [omni.physx.plugin] PhysX error: PxArticulationReducedCoordinate::updateKinematic(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, FILE C:\g\110116602\physx\source\physx\src\NpArticulationReducedCoordinate.cpp, LINE 684
2025-01-27 16:43:37 [22,256ms] [Warning] [omni.physx.plugin] Detected an articulation at /limo with more than 4 velocity iterations being added to a TGS scene.The related behavior changed recently, please consult the changelog. This warning will only print once.
2025-01-27 16:43:38 [22,357ms] [Error] [omni.physx.plugin] PhysX error: Fetching GPU Narrowphase failed! 700
, FILE C:\g\110116602\physx\source\gpunarrowphase\src\PxgNarrowphaseCore.cpp, LINE 1367
2025-01-27 16:43:38 [22,358ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-01-27 16:43:38 [22,358ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuEventRecord failed with error 700
, FILE C:\g\110116602\physx\source\gpucommon\include\PxgCudaUtils.h, LINE 57
2025-01-27 16:43:39 [23,616ms] [Warning] [omni.isaac.lab.sim.simulation_context] Simulation is stopped. The app will keep running with physics disabled. Press Ctrl+C or close the window to exit the app.
2025-01-27 16:43:39 [23,618ms] [Warning] [omni.physx.plugin] PhysX warning: C:\g\110116602\physx\source\gpucommon\src\PxgCudaMemoryAllocator.cpp, FILE C:\g\110116602\physx\source\gpucommon\src\PxgCudaMemoryAllocator.cpp, LINE 167
2025-01-27 16:43:39 [23,656ms] [Warning] [omni.physx.plugin] PhysX warning: CUDA context validation failed, FILE C:\g\110116602\physx\source\cudamanager\src\CudaContextManager.cpp, LINE 638
2025-01-27 16:43:39 [23,656ms] [Error] [omni.physx.plugin] Failed to create Cuda Context Manager.
2025-01-27 16:43:39 [23,657ms] [Error] [omni.physx.plugin] Unable to create PxCudaContextManager!
2025-01-27 16:43:39 [23,702ms] [Error] [carb.cudainterop.plugin] CUDA error 700: cudaErrorIllegalAddress - an illegal memory access was encountered)
2025-01-27 16:43:39 [23,703ms] [Error] [carb.cudainterop.plugin] Failed to import external memory in CUDA
2025-01-27 16:43:39 [23,703ms] [Error] [gpu.foundation.plugin] Cannot create cuda external memory for resource!
2025-01-27 16:43:39 [23,704ms] [Error] [gpu.foundation.plugin] Texture creation failed for the device: 0.
2025-01-27 16:43:39 [23,704ms] [Error] [carb.scenerenderer-rtx.plugin] Failed to allocate 1920x1080 LdrColor resource for device mask 0x1
2025-01-27 16:43:39 [23,705ms] [Warning] [omni.hydra] Rendering failed.
2025-01-27 16:43:39 [23,705ms] [Error] [omni.usd] HydraEngine::render failed to end the compute graph: error code 6
2025-01-27 16:43:39 [23,728ms] [Error] [carb.cudainterop.plugin] CUDA error 700: cudaErrorIllegalAddress - an illegal memory access was encountered)
2025-01-27 16:43:39 [23,729ms] [Error] [carb.cudainterop.plugin] Failed to import external memory in CUDA
[... 134 lines cut ...]
method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, FILE C:\g\110116602\physx\source\physx\src\NpArticulationReducedCoordinate.cpp, LINE 684   
2025-01-27 17:00:39 [29,534ms] [Warning] [omni.physx.plugin] Detected an articulation at /limo with more than 4 velocity iterations being added to a TGS scene.The related behavior changed recently, please consult the changelog. This warning will only print once.
  2%|█▉                                                       

```

### System Info

Describe the characteristic of your environment:

- Commit: -
- Isaac Sim Version: 4.2.0-rc.17+release.15988.c99988b1.gl
- OS: Windows 10
- GPU: RTX 3090
- CUDA: 12.7
- GPU Driver: 566.36

### Additional context

Other users have had the same errors, with different OS (Windows 11) and CUDA / GPU Drivers. Whilst I know that my GPU Driver is not the optimal for running Isaac Lab, I have had no problems with any scripts yet (almost tried every example).

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-01-28:
Thank you for submitting this. Could you confirm that you are not trying to add a robot to the scene while simulation is running? Scene setup changes should only happen before simulation starts. Thanks!

MuammerBay, 2025-01-29:
Hello Pablo, yes indeed I was trying to add a robot during simulation. I
made this issue because someone of the Omniverse Team suggested me to do
it, as it was "a bug nonetheless".
I personally modify the robot in Isaac Sim and then use the Scripts to
import these before simulation runs. I suppose that's the workflow and I
will feedforward the information!

Thanks!

MuammerBay, 2025-01-29:
Hello Pablo, yes indeed I was trying to add a robot during simulation. I made this issue because I got suggested to do it, as it was "a bug nonetheless". I personally modify the robot in Isaac Sim and then use the Scripts to import these before simulation runs. I suppose that's the workflow and I will feedforward the information!

Thanks!

RandomOakForest, 2025-03-07:
Ok, great. It seems you have resolved this issue. Feel free to reopen it if you find it persists in the latest versions of Isaac Lab and Isaac Sim. Thanks.

closed 2025-03-07 by RandomOakForest