isaac-sim/IsaacLab #1742: [Bug Report] Bug title pillow Heap-based Buffer Overflow  ( CWE-122 | CVE-2023-4863 | CVSS 9.6)

FarmerManUK24, 2025-01-28:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I ran the IsaacLab repository through Security Tool "SYNK".  and get a CVSS of 9.6 and CWE-122 for the folllowing:

https://security.snyk.io/vuln/SNYK-PYTHON-PILLOW-5918878

Here is the tool I use:  https://snyk.io/

### Steps to reproduce

I ran the above tool on the entire IsaacLab repository and get the above CVE-2023-4863.  

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Additional context

I am submitting given the very high CVE | CVSS score.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ]

[image]

[image]

- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

FarmerManUK24, 2025-01-28:
Understand some security tools may give false positives, but submitting, out of caution, given the High CVSS score of 9.6.

RandomOakForest, 2025-01-28:
Thank you for bringing this to our attention. By any chance you ran this on Isaac Sim 4.2 and Isaac Lab 1.4.0 and saw a similar score? This would be the current versions of the tool set. Thanks.

FarmerManUK24, 2025-01-28:
The Nvidia site below links to the Github repository I used:

https://developer.nvidia.com/isaac/lab

https://github.com/isaac-sim/IsaacLab

RandomOakForest, 2025-03-07:
Thank you for following up. The team has noted this to make the necessary improvements. Thanks.

closed 2025-03-07 by RandomOakForest