isaac-sim/IsaacLab #1745: Using multiple observation steps for each action step during RL Training.

ParlitsisG, 2025-01-28:
### Discussed in https://github.com/isaac-sim/IsaacLab/discussions/1744

<div type='discussions-op-text'>

<sup>Originally posted by **ParlitsisG** January 28, 2025</sup>
Hello, I am trying to make a Dual arm manipulation agent that performs actions with a frequency of (50Hz) but the agent uses a the output of a Neural Network that performs system identification using the applied torques and the pose of the manipulated object gathering samples at (500 hz).
 how could I incorporate this to train my rl agent using either a Manager based or direct workflow enviroment performing 10 simulation steps using a pd controller towards a fixed joint position gathering the required observation sequence and every 10th step using the rl agent giving the desired position of the joints.
 From what I understand I cant just progress the enviroment steps manually and apply the controller actions to get the observations inside the Direct workflow or ManagerbasedRL enviroment. 
 
 Any suggestions are welcome.</div>

KyleM73, 2025-01-30:
If you already have the low level network trained, you can use an action space or write an observation similar to what is done [here](https://github.com/isaac-sim/IsaacLab/blob/ea766783996d57e9f60b6c58c739c5fa4fbe79c5/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/manager_based/navigation/config/anymal_c/navigation_env_cfg.py#L49)

You could also refer to the image segmentation observation which uses a trained model for the observations [here](https://github.com/isaac-sim/IsaacLab/blob/ea766783996d57e9f60b6c58c739c5fa4fbe79c5/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/observations.py#L283)

RandomOakForest, 2025-01-30:
Thank you for adding this question. Let's follow up in #1744 under _Discussions_. Thanks.

closed 2025-01-30 by RandomOakForest