isaac-sim/IsaacLab #1747: [Bug Report] Potential decrease in runtime performance for release v.1.4.0

lukasfro, 2025-01-29:
### Describe the bug

I don't know whether you could call this a "bug" per se, but we noticed a decrease in runtime performance after upgrading to release v1.4.0. I'd be glad if someone else could confirm these results.

### Steps to reproduce

I ran the `benchmark_rlgames.py` script with tags `v1.3.0` and `v1.4.0` checked out respectively:
 
- `--task=Isaac-Cartpole-RGB-Camera-Direct-v0 --num_envs=512 --enable_cameras --headless`
- `--task=Isaac-Repose-Cube-Shadow-Direct-v0 --headless`
- `--task=Isaac-Humanoid-v0 --headless`

The following table summarizes some of the runtime KPIs that the benchmark provides:

For v1.3.0:

| in millisecs | `Isaac-Cartpole-RGB-Camera-Direct-v0` | `Isaac-Repose-Cube-Shadow-Direct-v0` | `Isaac-Humanoid-v0` |
|--------|--------|--------|--------|
| Min Env only step time | 0.9988 | 0.9332 | 0.4514 |
| Max Env only step time | 1.0786 | 2.4190 | 1.4377 |
| Mean Env only step time | 1.0267 | 1.1632 | 0.5798 | 

For v1.4.0:

| in millisecs | `Isaac-Cartpole-RGB-Camera-Direct-v0` | `Isaac-Repose-Cube-Shadow-Direct-v0` | `Isaac-Humanoid-v0` |
|--------|--------|--------|--------|
| Min Env only step time | 1.0561 | 0.9742 | 0.7952 |
| Max Env only step time | 1.4216 | 2.5009 | 1.6507 |
| Mean Env only step time | 1.1108 | 1.2365 | 0.9133 | 

Across all three tasks, the mean environment step time increases by up to 50% (Humanoid).

### System Info

Describe the characteristic of your environment:

- Commit: `v1.3.0` and `v1.4.0`
- Isaac Sim Version: `pip list` gives me `isaacsim 4.2.0.2`
- OS: Ubuntu 22.04
- GPU: RTX 4060 Ti
- CUDA: 12.2
- GPU Driver: 535.183.01

### Additional context

I'm happy to help out with more comparisons. 

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Mayankm96, 2025-01-30:
Yes, we are aware of this drop in the performance. The regression crept up in this MR: https://github.com/isaac-sim/IsaacLab/pull/966

We have made a fix since: https://github.com/isaac-sim/IsaacLab/pull/1695

@kellyguo11 @jtigue-bdai for vis

lukasfro, 2025-01-30:
Thanks for the quick reply @Mayankm96 I'm glad it is already fixed.

closed 2025-01-30 by Mayankm96