isaac-sim/IsaacLab #1756: [Bug Report] body_com_state_w not written to scene after reset

Jackkert, 2025-01-30:
### Describe the bug

Hi developers,
The Articulation class currently only has functions that write the root states to sim. So for example, upon reset, if configured, the reset_root_state_uniform function in omni/isaac/lab/envs/mdp/event.py currently calls these functions:
```asset.write_root_pose_to_sim(torch.cat([positions, orientations], dim=-1), env_ids=env_ids)```
```asset.write_root_velocity_to_sim(velocities, env_ids=env_ids)```

From my understanding, this writes the new root pose and velocity to the simulation and the ArticulationData buffer (in the case of an articulation) after the env has reset. However, this does not update the body_com_state_w buffer in ArticulationData. Meaning that after an environment has reset, the new root state is set, but the states of the other links is not. 

With the way how the ManagerBasedRLEnv (I'm referring to this function: https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py#L152) works, this means that the next time env.step() is called, the body_com_state_w is still the old buffer before the env was reset. The buffers are updated when scene.update(dt) is called, but that happens after the next time the policy is called in process_actions. This means that the action the policy outputs after an environment resets is based on the old body_com_state_w, for 1 environment step. In my case, my observations are retrieved from body_com_state_w, and so are the states later on in my control stack, so the agent will get a wrong reward signal based on the old body_com_state_w for 1 env step after reset. 

Is there a way to write body_com_state_w to the buffers right after the env has reset? Or update the scene of the reset envs separately after line 219?

### Steps to reproduce

- Get any ManagerBasedRLEnv
- In process_actions in the ActionTerm, log state for a link that is not the root link of your Articulation before the env is reset.
- Log the same state right after the env is reset for a link that is not the root link.
- The following states after that timestep should be updated after scene.update(dt) is called in https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py#L195

### System Info

Describe the characteristic of your environment:

- Commit: ea766783996d57e9f60b6c58c739c5fa4fbe79c5
- Isaac Sim Version: 4.2.0-rc.17+release.15988.c99988b1.gl%  
- OS: Ubuntu 22.04
- GPU: RTX 3090
- CUDA: 11.5
- GPU Driver: 535.183.01

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [x] body_com_state_w updates correctly after reset

Jackkert, 2025-01-30:
The issue

RandomOakForest, 2025-01-30:
Thank you for posting this. We will investigate.

Jackkert, 2025-01-31:
Suggestion:
I believe adding 
```self._data._body_link_state_w.timestamp = -1.0
self._data._body_com_state_w.timestamp = -1.0```

Below https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab/isaaclab/assets/articulation/articulation.py#L337
May resolve the inconsistent use of body_state_w and body_com_state_w

closed 2025-02-02 by Jackkert