isaac-sim/IsaacLab #1757: [Bug Report] Rsl_rl runner disregards action space when running DirectRLEnv based tasks

SuomiKP31, 2025-01-30:
### Describe the bug

When using DirectRLEnv with rsl_rl, its action output does not comply to the action space we defined.

### Steps to reproduce

For the quickest repro, we use the built-in Humanoid Direct task.
First, modify `direct_rl_env.py`, in the `_configure_gym_env_spaces` function, change:
```
self.single_action_space = gym.spaces.Box(low=-np.inf, high=np.inf, shape=(self.num_actions,))
```
to:
```
self.single_action_space = gym.spaces.Box(low=-1, high=1, shape=(self.num_actions,))
```

Then add a print statement in `locomotion_env.py`(which HumanoidDirectEnv inherits) to print out the actions returned from the RL agent.

**Run the task Isaac-Humanoid-Direct-v0:**
## Rsl_rl
Run:
```
./isaaclab.bat -p .\source\standalone\workflows\rsl_rl\train.py --task Isaac-Humanoid-Direct-v0 --num_envs 1024
```
Sample output:
```
tensor([[-0.3148, -0.2202, -0.1580,  ..., -0.7447, -0.6446,  0.8191],
        [-1.5024, -0.3462, -1.3022,  ..., -1.1339, -1.0219,  0.8803],
        [ 0.5253,  1.0459,  1.1591,  ...,  0.3404, -2.0113,  1.7017],
```
## rl_games
Run:
```
./isaaclab.bat -p .\source\standalone\workflows\rl_games\train.py --task Isaac-Humanoid-Direct-v0 --num_envs 1024
```

Sample output:
```
tensor([[ 1.0000, -0.5654,  0.1160,  ...,  0.1158, -0.3492, -1.0000],
        [-1.0000, -0.1038,  0.0163,  ..., -0.3761,  1.0000, -0.9950],
        [-0.0496,  0.0850,  0.7827,  ..., -0.7820,  0.1098,  0.5544],
        ...,
        [-0.5482, -0.0436,  1.0000,  ...,  0.0763, -0.7297,  0.9816],
        [-0.0203, -0.7350, -1.0000,  ..., -1.0000, -0.7373, -0.3905],
        [ 1.0000,  0.8091, -0.9549,  ...,  0.6334,  1.0000, -0.7724]],
       device='cuda:0')
```
We can notice that, rl_games clipped the action while rsl_rl didn't. 
I also tested skrl vs sb3, sb3 clipped the action while skrl didn't.

### System Info

Describe the characteristic of your environment:

Commit: https://github.com/isaac-sim/IsaacLab/commit/e00d62561a1b4ab19985bc15b40ac77c10bd5e33 (Release 1.20)
Isaac Sim Version: 4.2.0
OS: Win 11
GPU: RTX 4070 Ti S
CUDA: 12.6
GPU Driver: 566.03

### Additional context

While clipping can still be handled within your environment if you want a customized action space, as discussed in https://github.com/isaac-sim/IsaacLab/issues/1732, I would prefer the DirectRLEnv class to have clipping function regardless of the rl agent we run, because it is possible that task-based actions (such as phase modulation) have a different space (e.g. [0, dt]). 

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] rsl_rl and skrl runners clip their actions returned to the environment to limits set by `self.action_space` correctly.

Mayankm96, 2025-01-30:
This seems like something the libraries don't handle.

In RSL-RL at least we don't do anything with the Gym-related spaces that the environment provides. So I would expect this to be the behavior.

RandomOakForest, 2025-01-31:
Thank you for posting this. As @Mayankm96 points out, please follow through with issues in those repositories. Let us know if we can be of further help.

closed 2025-01-31 by RandomOakForest