isaac-sim/IsaacLab #1760: [Bug] Urdf import not working

YannickP99, 2025-01-31:
### Describe the bug

My code to import an URDF file into my DirectRLEnvCfg stopped working with the new IsaacSim update.

### Steps to reproduce
This is my code:
```
@configclass
class Go2EnvCfg(DirectRLEnvCfg):
    #* Simulation
    sim: SimulationCfg = SimulationCfg(
        dt=1 / 200,
        render_interval=decimation,
        disable_contact_processing=True,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
    )

    #* Terrain
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="plane",
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
        debug_vis=False,
    )

    #* Scene
    scene: InteractiveSceneCfg = InteractiveSceneCfg(num_envs=4096, env_spacing=5.0, replicate_physics=False)

    #* Robot
    robot = ArticulationCfg(
        prim_path="/World/envs/env_.*/Robot",
        spawn=sim_utils.UrdfFileCfg(
            asset_path="source/thesis/resources/go2_manip_description/urdf/go2_manip_comb.urdf",
            usd_dir=f"source/thesis/resources/go2_manip_description/usd",
            usd_file_name="go2_manip_exported",
            fix_base=False,
            force_usd_conversion=True,
            activate_contact_sensors=True,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                disable_gravity=False,
                retain_accelerations=False,
                linear_damping=0.0,
                angular_damping=0.0,
                max_linear_velocity=1000.0,
                max_angular_velocity=1000.0,
                max_depenetration_velocity=1.0,
            ),
            articulation_props=sim_utils.ArticulationRootPropertiesCfg(
                enabled_self_collisions=False, solver_position_iteration_count=4, solver_velocity_iteration_count=0
            ),
            joint_drive=UrdfConverterCfg.JointDriveCfg(
                gains=UrdfConverterCfg.JointDriveCfg.PDGainsCfg(stiffness=None, damping=None) #stiffness=100.0, damping=1.0
            ),
        ),
        init_state=ArticulationCfg.InitialStateCfg(
            pos=(0.0, 0.0, 0.4),
            joint_pos={
                ".*L_hip_joint": 0.1,
                ".*R_hip_joint": -0.1,
                "F[L,R]_thigh_joint": 0.8,
                "R[L,R]_thigh_joint": 1.0,
                ".*_calf_joint": -1.5,
                "joint_connector_gripper": 0.0,
            },
            joint_vel={".*": 0.0},
        ),
        soft_joint_pos_limit_factor=0.9,
        actuators={
            "base_legs": DCMotorCfg(
                joint_names_expr=[".*_hip_joint", ".*_thigh_joint", ".*_calf_joint"],
                effort_limit=23.5,
                saturation_effort=23.5,
                velocity_limit=30.0,
                stiffness=25.0,
                damping=0.5,
                friction=0.0,
            ),
            # TODO: get real attribute information
            "base_arm": DCMotorCfg(
                joint_names_expr=["joint_connector_gripper"],
                effort_limit=5.0,
                saturation_effort=5.0,
                velocity_limit=10.0,
                stiffness=3.0,
                damping=0.5,
                friction=0.0,
            ),
        },
    )
```
Error Code (added printing statements to "sim.schemas.schemas.py activate_contact_sensors-function" to understand the prims and it looks like only one prim of the robot is detected):

```
2025-01-31 06:10:44 [8,121ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/envs/env_0/Robot>: Could not open asset @configuration/go2_manip_exported_physics.usd@ for payload introduced by @source/thesis/resources/go2_manip_description/usd/go2_manip_exported.usd@</go_manip_description{Physics=PhysX}>. (recomposing stage on stage @anon:0x25529420:World0.usd@ <0x28175460>)

2025-01-31 06:10:44 [8,121ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/envs/env_0/Robot>: Could not open asset @configuration/go2_manip_exported_sensor.usd@ for payload introduced by @source/thesis/resources/go2_manip_description/usd/go2_manip_exported.usd@</go_manip_description{Sensor=Sensors}>. (recomposing stage on stage @anon:0x25529420:World0.usd@ <0x28175460>)

2025-01-31 06:10:44 [8,162ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_rigid_body_properties' on any prims under: '/World/envs/env_0/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
2025-01-31 06:10:44 [8,162ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_articulation_root_properties' on any prims under: '/World/envs/env_0/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
['/World/envs/env_0/Robot', '/World/envs/env_1/Robot', '/World/envs/env_2/Robot', '/World/envs/env_3/Robot']
all_prims: [Usd.Prim(</World/envs/env_0/Robot>)]
child_prim: []
[INFO]: Time taken for scene creation : 0.773408 seconds
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/yannick/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/yannick/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 105, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "/home/yannick/anaconda3/envs/isaaclab_new/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 740, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/yannick/IsaacLab/source/thesis/thesis/quadruped_direct/go2_env.py", line 27, in __init__
    super().__init__(cfg, render_mode, **kwargs)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/envs/direct_rl_env.py", line 123, in __init__
    self._setup_scene()
  File "/home/yannick/IsaacLab/source/thesis/thesis/quadruped_direct/go2_env.py", line 106, in _setup_scene
    self._robot = Articulation(self.cfg.robot)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 99, in __init__
    super().__init__(cfg)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 77, in __init__
    self.cfg.spawn.func(
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/sim/utils.py", line 277, in wrapper
    schemas.activate_contact_sensors(prim_paths[0], cfg.activate_contact_sensors)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/sim/schemas/schemas.py", line 519, in activate_contact_sensors
    raise ValueError(
ValueError: No contact sensors added to the prim: '/World/envs/env_0/Robot'. This means that no rigid bodies are present under this prim. Please check the prim path.
```
I also tried to import the URDF into IsaacSim using the UI and saw that now also some "configurations"-folder is created in the USD export folder. When visualising the different configurations I found out that "_base" and "_physics" are visualized normally but the "_sensors" looks empty.

### System Info

Describe the characteristic of your environment:
- Commit: 157a19e
- Isaac Sim Version: 4.5.0.0
- OS: Ubuntu 22.04.5 LTS
- GPU: RTX 3090
- CUDA: 12.4
- GPU Driver: 550.120

RandomOakForest, 2025-01-31:
Thank you for posting this. We'll take a look.

YannickP99, 2025-02-01:
I think it is a more general problem. I am also trying to spawn an USD object which I imported from a stl-file and I get a similar error:
```
['/World/envs/env_0/bin_small', '/World/envs/env_1/bin_small', '/World/envs/env_2/bin_small', '/World/envs/env_3/bin_small']
2025-02-01 00:12:03 [8,679ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_rigid_body_properties' on any prims under: '/World/envs/env_0/bin_big'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
['/World/envs/env_0/bin_big', '/World/envs/env_1/bin_big', '/World/envs/env_2/bin_big', '/World/envs/env_3/bin_big']
['/World/ground/physicsMaterial']
['/World/Light']
['/World/Markers/marker_bin_small/sphere/geometry/material']
['/World/Markers/marker_bin_small/sphere']
['/World/Markers/marker_bin_big/sphere/geometry/material']
['/World/Markers/marker_bin_big/sphere']
2025-02-01 00:12:03 [8,995ms] [Warning] [isaaclab.scene.interactive_scene] Collision filtering can only be automatically enabled when replicate_physics=True. Please call scene.filter_collisions(global_prim_paths) to filter collisions across environments.
[INFO]: Time taken for scene creation : 1.794014 seconds
[INFO]: Scene manager:  <class InteractiveScene>
	Number of environments: 4
	Environment spacing   : 5.0
	Source prim name      : /World/envs/env_0
	Global prim paths     : ['/World/ground']
	Replicate physics     : False
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
Traceback (most recent call last):
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 94, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 265, in _initialize_callback
    self._initialize_impl()
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/rigid_object/rigid_object.py", line 417, in _initialize_impl
    raise RuntimeError(
RuntimeError: Failed to find a rigid body when resolving '/World/envs/env_.*/bin_small'. Please ensure that the prim has 'USD RigidBodyAPI' applied.
Traceback (most recent call last):
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 94, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 265, in _initialize_callback
    self._initialize_impl()
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/rigid_object/rigid_object.py", line 417, in _initialize_impl
    raise RuntimeError(
RuntimeError: Failed to find a rigid body when resolving '/World/envs/env_.*/bin_big'. Please ensure that the prim has 'USD RigidBodyAPI' applied.
2025-02-01 00:12:05 [10,592ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /World/envs/env_0/Robot/FR_thigh/visuals.proto_mesh_1_id1 that has not been populated
[INFO]: Time taken for simulation start : 1.759270 seconds
[INFO] Event Manager:  <EventManager> contains 2 active terms.
+--------------------------------------+
| Active Event Terms in Mode: 'startup' |
+----------+---------------------------+
|  Index   | Name                      |
+----------+---------------------------+
|    0     | physics_material          |
|    1     | add_base_mass             |
+----------+---------------------------+
+-------------------------------------+
| Active Event Terms in Mode: 'reset' |
+---------+---------------------------+
|  Index  | Name                      |
+---------+---------------------------+
|    0    | reset_robot_base          |
|    1    | reset_bin_small           |
|    2    | reset_bin_big             |
|    3    | reset_robot_joints        |
+---------+---------------------------+

Creating window for environment.
ManagerLiveVisualizer cannot be created for manager: action_manager, Manager does not exist
ManagerLiveVisualizer cannot be created for manager: observation_manager, Manager does not exist
[INFO]: Completed setting up the environment...
> Target link 'grasping_point_link' at id 0
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/yannick/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/yannick/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 129, in main
    env = RslRlVecEnvWrapper(env)
  File "/home/yannick/IsaacLab/source/isaaclab_rl/isaaclab_rl/rsl_rl/vecenv_wrapper.py", line 89, in __init__
    self.env.reset()
  File "/home/yannick/anaconda3/envs/isaaclab_new/lib/python3.10/site-packages/gymnasium/wrappers/common.py", line 400, in reset
    return super().reset(seed=seed, options=options)
  File "/home/yannick/anaconda3/envs/isaaclab_new/lib/python3.10/site-packages/gymnasium/core.py", line 328, in reset
    return self.env.reset(seed=seed, options=options)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/envs/direct_rl_env.py", line 275, in reset
    self._reset_idx(indices)
  File "/home/yannick/IsaacLab/source/thesis/thesis/quadruped_direct/go2_env.py", line 371, in _reset_idx
    super()._reset_idx(env_ids)                 # type: ignore
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/envs/direct_rl_env.py", line 565, in _reset_idx
    self.scene.reset(env_ids)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/scene/interactive_scene.py", line 425, in reset
    rigid_object.reset(env_ids)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/rigid_object/rigid_object.py", line 103, in reset
    self._external_force_b[env_ids] = 0.0
AttributeError: 'RigidObject' object has no attribute '_external_force_b'
```

linchangyi1, 2025-02-05:
Hi @YannickP99 and @RandomOakForest, I also met this issue and would like to kindly ask if you have fixed it?

YannickP99, 2025-02-06:
Hey @linchangyi1, unfortunately I wasnt able to fix it so I migrated back to the previous version

linchangyi1, 2025-02-06:
Thanks for your reply @YannickP99, I found a similar issue posted here https://github.com/isaac-sim/IsaacLab/issues/1688 
But I didn't get how it was fixed.

linchangyi1, 2025-02-06:
Thanks for your reply @YannickP99

linchangyi1, 2025-02-06:
@RandomOakForest Could you please take a look at this bug~ Thanks a lot!

linchangyi1, 2025-02-06:
It seems this bug was reported (https://github.com/isaac-sim/IsaacLab/issues/1536) last year

YannickP99, 2025-02-06:
Oh okay. I am using commit ea766783 from the 27th January and my code is working fine.

kellyguo11, 2025-02-06:
Are you able to share the URDF file you are having issues with? It could be the new URDF importer had some modifications in the paths where the schemas are being applied to.

YannickP99, 2025-02-06:
yes sure here is the urdf. In case you need the whole folder, i can share it with you too,
```
<?xml version="1.0" encoding="utf-8"?>
<!-- This URDF was automatically created by SolidWorks to URDF Exporter! Originally created by
Stephen Brawner (brawner@gmail.com) 
     Commit Version: 1.6.0-4-g7f85cfe  Build Version: 1.6.7995.38578
     For more information, please see http://wiki.ros.org/sw_urdf_exporter -->
<robot name="go_manip_description">
  <link name="base">
    <inertial>
      <origin xyz="0.021112 0 -0.005366" rpy="0 0 0" />
      <mass value="6.921" />
      <inertia ixx="0.02448" ixy="0.00012166" ixz="0.0014849" iyy="0.098077" iyz="-3.12E-05" izz="0.107" />
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
          <mesh filename="../dae/base.dae" /> 
      </geometry>
      <material name="">
        <color rgba="1 1 1 1" />
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0" />
      <geometry>
        <box size="0.3762 0.0935 0.114" />
      </geometry>
    </collision>
  </link>
  <link name="Head_upper">
    <inertial>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <mass value="0.001" />
      <inertia ixx="9.6e-06" ixy="0" ixz="0" iyy="9.6e-06" iyz="0" izz="9.6e-06" />
    </inertial>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <cylinder radius="0.05" length="0.09" />
      </geometry>
    </collision>
  </link>
  <joint name="Head_upper_joint" type="fixed" dont_collapse="true">
    <origin xyz="0.285 0 0.01" rpy="0 0 0" />
    <parent link="base" />
    <child link="Head_upper" />
    <axis xyz="1 0 0" />
    </joint>
  <!-- <link name="Head_lower">
    <inertial>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <mass value="0.001" />
      <inertia ixx="9.6e-06" ixy="0" ixz="0" iyy="9.6e-06" iyz="0" izz="9.6e-06" />
    </inertial>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <sphere radius="0.047" />
      </geometry>
    </collision>
  </link>
  <joint name="Head_lower_joint"
    type="fixed" dont_collapse="true">
    <origin xyz="0.008 0 -0.07" rpy="0 0 0" />
    <parent link="Head_upper" />
    <child link="Head_lower" />
    <axis xyz="1 0 0" />
  </joint> -->
  <link name="FL_hip">
[... 681 lines cut ...]
    <!-- <visual>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry>
        <sphere radius="0.001"/>
      </geometry>
      <material name="">
        <color rgba="1 0 0 1" />
      </material>
    </visual> -->
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry>
        <empty/>
      </geometry>
    </collision>
  </link>
  <joint name="joint_grasping_point" type="fixed">
    <parent link="link3_gripper"/>
    <child link="grasping_point_link"/>
    <origin xyz="-0.005 0.163 0" rpy="0 0 0"/>
      <axis xyz="0 1 0"/>
      <limit lower="0" upper="0" />
  </joint>
</robot>
```

linchangyi1, 2025-02-06:
@kellyguo11 I suspect that the issue in the URDF converter mentioned here (https://github.com/isaac-sim/IsaacLab/issues/1800) likely causes the following problem when processing the USD in IsaacLab.

vicltz, 2025-02-10:
I have a similar issue.
I just ran the conversion script that returned the following error:

```
2025-02-10 09:10:08 [36,489ms] [Error] [omni.kit.commands.command] Failed to execute a command: URDFImportRobot.
  File "/home/myname/isaac-lab-workspace/IsaacLab/scripts/tools/convert_urdf.py", line 134, in <module>
    main()
  File "/home/myname/isaac-lab-workspace/IsaacLab/scripts/tools/convert_urdf.py", line 104, in main
    urdf_converter = UrdfConverter(urdf_converter_cfg)
  File "/home/myname/isaac-lab-workspace/IsaacLab/source/isaaclab/isaaclab/sim/converters/urdf_converter.py", line 55, in __init__
    super().__init__(cfg=cfg)
  File "/home/myname/isaac-lab-workspace/IsaacLab/source/isaaclab/isaaclab/sim/converters/asset_converter_base.py", line 106, in __init__
    self._convert_asset(cfg)
  File "/home/myname/isaac-lab-workspace/IsaacLab/source/isaaclab/isaaclab/sim/converters/urdf_converter.py", line 84, in _convert_asset
    omni.kit.commands.execute(
  File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/commands/command.py", line 463, in execute
    result = omni.kit.undo.execute(command, name, kwargs)
[...skipped...]
  File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 78, in execute
    result = _execute(command, name, level, history_key)
  File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 459, in _execute
    raise error
  File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 420, in _execute
    result = command.do()
  File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/isaacsim.asset.importer.urdf-2.3.10+106.4.0.lx64.r.cp310/isaacsim/asset/importer/urdf/scripts/commands.py", line 145, in do
    return self._urdf_interface.import_robot(

 <class 'RuntimeError'> Used null prim

```

Ycyofmine, 2025-04-07:
Solution for Ubuntu 22.04 users:
1. Delete the entire ov folder located at ~/.local/share/ov.
2. Re-run the Isaac Sim selector script:
```shell
./isaacsim-selector.sh
```
This should resolve the URDF import issue.

vicltz, 2025-04-23:
> Solution for Ubuntu 22.04 users:
> 
>     1. Delete the entire ov folder located at ~/.local/share/ov.
> 
>     2. Re-run the Isaac Sim selector script:
> 
> 
> ./isaacsim-selector.sh
> 
> This should resolve the URDF import issue.

Where is this `isaacsim-selector.sh` script located ?

YannickP99, 2025-05-04:
@Ycyofmine @RandomOakForest I am still having this problem. I would be very thankful if you could take another look. I am using commit: 70a5ffcdda943fe6b6f450379929a0a9daab4e0b
This is the error message:
```
[INFO]: Base environment:
	Environment device    : cuda:0
	Environment seed      : 42
	Physics step-size     : 0.005
	Rendering step-size   : 0.02
	Environment step-size : 0.02
2025-05-04 07:15:18 [3,953ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/envs/env_0/Robot>: Could not open asset @configuration/usd_3_physics.usd@ for payload introduced by @source/thesis/resources/go2_manip_description/usd_3/usd_3.usd@</go_manip_description{Physics=PhysX}>. (recomposing stage on stage @anon:0x10ebbd00:World0.usd@ <0x10ec0b50>)

2025-05-04 07:15:18 [3,953ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/envs/env_0/Robot>: Could not open asset @configuration/usd_3_sensor.usd@ for payload introduced by @source/thesis/resources/go2_manip_description/usd_3/usd_3.usd@</go_manip_description{Sensor=Sensors}>. (recomposing stage on stage @anon:0x10ebbd00:World0.usd@ <0x10ec0b50>)

2025-05-04 07:15:18 [3,997ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_rigid_body_properties' on any prims under: '/World/envs/env_0/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
2025-05-04 07:15:18 [3,997ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_articulation_root_properties' on any prims under: '/World/envs/env_0/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
2025-05-04 07:15:18 [3,997ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_joint_drive_properties' on any prims under: '/World/envs/env_0/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
[INFO]: Time taken for scene creation : 0.046341 seconds
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/yannick/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/yannick/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 183, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "/home/yannick/anaconda3/envs/isaaclab_new2/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 740, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/yannick/IsaacLab/source/thesis/thesis/quadruped_wall/go2_wall_asym_actorcritic.py", line 32, in __init__
    super().__init__(cfg, render_mode, **kwargs)
  File "/home/yannick/IsaacLab/source/thesis/thesis/quadruped_wall/go2_wall_env.py", line 30, in __init__
    super().__init__(cfg, render_mode, **kwargs)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/envs/direct_rl_env.py", line 127, in __init__
    self._setup_scene()
  File "/home/yannick/IsaacLab/source/thesis/thesis/quadruped_wall/go2_wall_asym_actorcritic.py", line 37, in _setup_scene
    super()._setup_scene()
  File "/home/yannick/IsaacLab/source/thesis/thesis/quadruped_wall/go2_wall_env.py", line 128, in _setup_scene
    self._robot = Articulation(self.cfg.robot)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 100, in __init__
    super().__init__(cfg)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 78, in __init__
    self.cfg.spawn.func(
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/sim/utils.py", line 276, in wrapper
    schemas.activate_contact_sensors(prim_paths[0], cfg.activate_contact_sensors)
  File "/home/yannick/IsaacLab/source/isaaclab/isaaclab/sim/schemas/schemas.py", line 519, in activate_contact_sensors
    raise ValueError(
ValueError: No contact sensors added to the prim: '/World/envs/env_0/Robot'. This means that no rigid bodies are present under this prim. Please check the prim path.
```
I am importing the USD using:
```
go2_gripper = ArticulationCfg(
        prim_path="/World/envs/env_.*/Robot",
        spawn=sim_utils.UsdFileCfg(
            activate_contact_sensors=True,
            usd_path="source/thesis/resources/go2_manip_description/usd_3/usd_3.usd",
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                disable_gravity=False,
                retain_accelerations=True,
                linear_damping=0.0,
                angular_damping=0.0,
                max_linear_velocity=1000.0,
                max_angular_velocity=1000.0,
                max_depenetration_velocity=1000.0
            ),
[... 50 lines cut ...]
                friction=0.0,
            ),
            "base_arm": DCMotorCfg(
                joint_names_expr=["joint_connector_gripper"],
                effort_limit_sim=1.4, 
                effort_limit=1.4,
                saturation_effort=1.4,
                velocity_limit=5.0,
[... cut ...] ess=3.0,
                damping=0.5,
                friction=0.0,
            ),
        },
    )
```
This is an exported folder containing all URDF data and the exported USD file:
[urdf_folder.zip](https://github.com/user-attachments/files/20026512/urdf_folder.zip)

I did two screenshots in IsaacSim which show a different structure in my old (working) and new (not working) USD file. Is this difference correct?

Old model which I exported at 6th february:
[image]

Current model export:
[image]

kellyguo11, 2025-05-11:
Can you make sure you are running the latest URDF Importer, which should be version 2.3.14? The version check and update can be done through the Extension Manager in Isaac Sim.

YannickP99, 2025-05-11:
@kellyguo11 thank you for the response. I updated now to 2.3.14 but it is still not working. I also tried to add the option root_link_name to the UrdfConverterCfg in convert_urdf.py, but it does not help

YannickP99, 2025-05-20:
@kellyguo11 do you have any further advices for me?

kellyguo11, 2025-05-20:
Did it work for you if you used the importer directly from the Isaac Sim GUI? trying to narrow down if it's a bug in the Isaac Lab wrapper or in the importer.

YannickP99, 2025-05-20:
Using both variants does not give me a direct error but when I want to use the transformed USD model in my IsaacLab code, it returns the "No contact sensor"-error

aravindev, 2025-05-28:
> > I have a similar issue. I just ran the conversion script that returned the following error:
> > ```
> > 2025-02-10 09:10:08 [36,489ms] [Error] [omni.kit.commands.command] Failed to execute a command: URDFImportRobot.
> >   File "/home/myname/isaac-lab-workspace/IsaacLab/scripts/tools/convert_urdf.py", line 134, in <module>
> >     main()
> >   File "/home/myname/isaac-lab-workspace/IsaacLab/scripts/tools/convert_urdf.py", line 104, in main
> >     urdf_converter = UrdfConverter(urdf_converter_cfg)
> >   File "/home/myname/isaac-lab-workspace/IsaacLab/source/isaaclab/isaaclab/sim/converters/urdf_converter.py", line 55, in __init__
> >     super().__init__(cfg=cfg)
> >   File "/home/myname/isaac-lab-workspace/IsaacLab/source/isaaclab/isaaclab/sim/converters/asset_converter_base.py", line 106, in __init__
> >     self._convert_asset(cfg)
> >   File "/home/myname/isaac-lab-workspace/IsaacLab/source/isaaclab/isaaclab/sim/converters/urdf_converter.py", line 84, in _convert_asset
> >     omni.kit.commands.execute(
> >   File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/commands/command.py", line 463, in execute
> >     result = omni.kit.undo.execute(command, name, kwargs)
> > [...skipped...]
> >   File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 78, in execute
> >     result = _execute(command, name, level, history_key)
> >   File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 459, in _execute
> >     raise error
> >   File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/omni.kit.commands-1.4.9+d02c707b.lx64.r.cp310/omni/kit/undo/undo.py", line 420, in _execute
> >     result = command.do()
> >   File "/home/myname/miniforge3/envs/isaaclab2/lib/python3.10/site-packages/isaacsim/extscache/isaacsim.asset.importer.urdf-2.3.10+106.4.0.lx64.r.cp310/isaacsim/asset/importer/urdf/scripts/commands.py", line 145, in do
> >     return self._urdf_interface.import_robot(
> > 
> >  <class 'RuntimeError'> Used null prim
> > ```
> 
> I was able to track this error down to an stl file which the importer did not like. If I reuse an stl file from a previous link this error is gone. All in all the import from urdf is really hard, I would rather go the detour via mujoco xml files.
> 
> **I can confirm that my urdf was perfectly imported in isaacsim 4.2! For now I would avoid the isaacsim 4.5 importer and rather take the detour to IsaacSim 4.2!**

Had a very similar issue today while I was trying to convert a URDF file to USD. Although the DAE fliles used for meshes work fine in with IsaacSim 4.2, it somehow errors out in IsaacSim 4.5

```
2025-05-28 15:01:36 [13,749ms] [Warning] [omni.usd] Warning: in SdfPath at line 81 of /builds/omniverse/usd-ci/USD/pxr/usd/sdf/path.cpp -- Ill-formed SdfPath </visuals/dynaarm_SHOULDER/100_s
houlder_mesh>: syntax error                                                                                                                                                                   
                                                                                                                                                                                              
2025-05-28 15:01:36 [13,749ms] [Warning] [omni.usd] Warning: in SdfPath at line 81 of /builds/omniverse/usd-ci/USD/pxr/usd/sdf/path.cpp -- Ill-formed SdfPath </visuals/dynaarm_SHOULDER/100_s
houlder_mesh>: syntax error                                                                                                                                                                   
                                                                                                                                                                                              
2025-05-28 15:01:36 [13,749ms] [Warning] [omni.usd] Coding Error: in _IsValidPathForCreatingPrim at line 3340 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- Path must be an absolut
e path: <>                                                                                                                                                                                    
                                                                                                                                                                                              
2025-05-28 15:01:36 [13,751ms] [Error] [omni.kit.commands.command] Failed to execute a command: URDFImportRobot.                                                                              
  File "/IsaacLab-Internal/scripts/tools/convert_urdf.py", line 161, in <module>                                                                            
    main()                                                                                                                                                                                    
  File "/IsaacLab-Internal/scripts/tools/convert_urdf.py", line 131, in main                                                                                
[... cut ...]
                                                                                                                                                                                              
 <class 'RuntimeError'> Used null prim
```

Interesting thing is the warning here: `2025-05-28 15:01:36 [13,749ms] [Warning] [omni.usd] Warning: in SdfPath at line 81 of /builds/omniverse/usd-ci/USD/pxr/usd/sdf/path.cpp -- Ill-formed SdfPath </visuals/dynaarm_SHOULDER/100_s
houlder_mesh>: syntax error` which does not show up with the previous version.

markcfong561, 2025-06-12:
Hi, what is the status on this issue?

markcfong561, 2025-06-13:
that doesn't necessarily help since the urdf importer on isaac sim itself doesn't work

fhndan89, 2025-06-13:
Ah okay. I use the urdf-converter from IsaacLab.

ooctipus, 2025-07-08:
Can you confirm you IsaacLab version?
Mine is IsaacLab2.2

I had no problem using either below spawner cfg to convert directly from urdf in the file you share to the usd. With contact sensor all working.
```
spawn=sim_utils.UrdfFileCfg(
            asset_path="/.../go2_manip_description/urdf/go2_manip_comb.urdf",
            usd_dir=f"/.../go2_manip_description/usd",
            usd_file_name="go2_manip_comb.usd",
            fix_base=False,
            merge_fixed_joints=False,
            force_usd_conversion=True,
            activate_contact_sensors=True,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                disable_gravity=False,
                retain_accelerations=False,
                linear_damping=0.0,
                angular_damping=0.0,
                max_linear_velocity=1000.0,
                max_angular_velocity=1000.0,
                max_depenetration_velocity=1.0,
            ),
            articulation_props=sim_utils.ArticulationRootPropertiesCfg(
                enabled_self_collisions=False, solver_position_iteration_count=4, solver_velocity_iteration_count=0
            ),
            joint_drive=UrdfConverterCfg.JointDriveCfg(
                gains=UrdfConverterCfg.JointDriveCfg.PDGainsCfg(stiffness=100, damping=1.) #stiffness=100.0, damping=1.0
            ),
        ),
```

Another way I also tested has no problem was
`python script/tools/convert_urdf.py /.../go2_manip_description/urdf/go2_manip_comb.urdf  /.../go2_manip_description/usd/go2_manip_comb.usd`

then replace above code sim_utils.UrdfFileCfg with sim_utils.UsdFileCfg

Maybe can you try either approach in the most up to date IsaacLab and see if issue persists?

ooctipus, 2025-07-10:
We tried several different URDF import cases by now in our end, it is working fine in both isaaclab2.1 + sim4.5, isaaclab2.2 + sim5.0 , if you are not experience issue anymore maybe we can close the issue for now. If someday you see it again, feel free to reopen.

closed 2025-07-10 by ooctipus

johbrust, 2026-02-04:
I had a similar issue, which was somehow caused by the file names of the meshes (e.g., "331839-DUMMY-M.stl"). I renamed the files to simpler ones (e.g., "bracket.stl"), which fixed the issue. I tried to make sense of that, but was not able to identify the reason. Maybe this insight helps someone with a similar problem.