isaac-sim/IsaacLab #1763: [Bug Report] 2.0 Rendered Scrips Crash on Launchg

ramzibou0735, 2025-01-31:
### Describe the bug
As I was filling this bug report, I tried to reproduce it but it seems like everything is back to normal, but I thought about posting it anyways if it could help the maintainers or anyone facing the same issue
After updating the Isaac Sim pip installation to version 4.5, making sure that it works, updating the IsaacLab repo to the current main branch:
Any script launched that isn't --headless spawns in an app window that will stay frozen until it crashes, this goes for tutorial scripts, agent training and demo scripts.

### Steps to reproduce

After finishing all the installation guides here https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html :

```
conda activate isaaclab
cd Isaaclab
./isaaclab.sh -p /scripts/demos/quadcopter.py

[6.212s] app ready
2025-01-31 07:42:14 [6,220ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 7374781440
2025-01-31 07:42:14 [6,220ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid false, within: false
2025-01-31 07:42:14 [6,220ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : decrement: 167690, decrement size: 7301033856
2025-01-31 07:42:14 [6,220ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : New limit 9748724 (slope: 439, intercept: 13179904)
2025-01-31 07:42:14 [6,220ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 4287352704
2025-01-31 07:42:14 [6,220ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid true, within: true
2025-01-31 07:42:14 [6,371ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults-transient/meshlights/forceDisable'
[9.716s] Simulation App Startup Complete
[9.810s] [ext: omni.physx.fabric-106.5.7] startup
2025-01-31 07:44:51 [163,503ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/Crazyflie>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/Robots/Crazyflie/cf2x.usd@ for reference introduced by @anon:0x414e8740:World0.usd@</World/Crazyflie>. (recomposing stage on stage @anon:0x414e8740:World0.usd@ <0x43d6e2c0>)

2025-01-31 07:44:51 [163,523ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_rigid_body_properties' on any prims under: '/World/Crazyflie'. This might be because of the following reasons:
(1) The desired attribute does not exist on any of the prims.
(2) The desired attribute exists on an instanced prim.
Discovered list of instanced prim paths: []
2025-01-31 07:44:51 [163,523ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_articulation_root_properties' on any prims under: '/World/Crazyflie'. This might be because of the following reasons:
(1) The desired attribute does not exist on any of the prims.
(2) The desired attribute exists on an instanced prim.
Discovered list of instanced prim paths: []
2025-01-31 07:44:51 [163,524ms] [Warning] [isaaclab.sim.spawners.from_files.from_files] A prim already exists at prim path: '/World/Crazyflie'.
2025-01-31 07:44:51 [163,524ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_rigid_body_properties' on any prims under: '/World/Crazyflie'. This might be because of the following reasons:
(1) The desired attribute does not exist on any of the prims.
(2) The desired attribute exists on an instanced prim.
Discovered list of instanced prim paths: []
2025-01-31 07:44:51 [163,524ms] [Warning] [isaaclab.sim.utils] Could not perform 'modify_articulation_root_properties' on any prims under: '/World/Crazyflie'. This might be because of the following reasons:
(1) The desired attribute does not exist on any of the prims.
(2) The desired attribute exists on an instanced prim.
Discovered list of instanced prim paths: []
Traceback (most recent call last):
File "/home/p/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 94, in <lambda>
lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
File "/home/p/IsaacLab/source/isaaclab/isaaclab/assets/asset_base.py", line 265, in _initialize_callback
self._initialize_impl()
File "/home/p/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 1094, in _initialize_impl
raise RuntimeError(
RuntimeError: Failed to find an articulation when resolving '/World/Crazyflie'. Please ensure that the prim has 'USD ArticulationRootAPI' applied.
Traceback (most recent call last):
File "/home/p/IsaacLab/scripts/demos/quadcopter.py", line 116, in <module>
main()
File "/home/p/IsaacLab/scripts/demos/quadcopter.py", line 73, in main
prop_body_ids = robot.find_bodies("m.*_prop")[0]
File "/home/p/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 221, in find_bodies
return string_utils.resolve_matching_names(name_keys, self.body_names, preserve_order)
File "/home/p/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 146, in body_names
return self.root_physx_view.shared_metatype.link_names
File "/home/p/IsaacLab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 155, in root_physx_view
return self._root_physx_view
AttributeError: 'Articulation' object has no attribute '_root_physx_view'. Did you mean: 'root_physx_view'?
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-01-31 07:44:59 [171,346ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2025-01-31 07:44:59 [171,348ms] [Warning] [omni.graph.core.plugin] Could not find category 'animation' for removal
2025-01-31 07:44:59 [171,470ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-01-31 07:44:59 [171,470ms] [Warning] [carb.audio.context] 1 contexts were leaked
2025-01-31 07:44:59 [171,490ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

### System Info

Describe the characteristic of your environment:

- Commit: 157a19e29b0dcafd87abbdad6f2e9428df3f09be
- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU: RTX 4070 Super
- CUDA: 12.4
- GPU Driver: 560.35.05

### Additional context

Add any other context about the problem here.

### Checklist

- [ ✅] I have checked that there is no similar issue in the repo (**required**)
- [ ✅] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-01-31:
Thank you for posting this. Let us know if you see such crash again. Thanks.

itsmehere, 2025-02-02:
@ramzibou0735 I'm having the same exact issue. Do you remember what in particular might have fixed it, or did you just run it again and everything worked? It was fine for me before the Isaac sim/lab update, but now I'm getting a `RuntimeError: Failed to find an articulation when resolving '[...]'. Please ensure that the prim has 'USD ArticulationRootAPI' applied` error.

ramzibou0735, 2025-02-02:
@itsmehere can't really put my finger on what made it work for me, when it used to crash I launched all the examples available on the repo for hours but as long as it wasn't running headless it would crash, but when I was writing the bug report and wanted to run the easiest script to reproduce those errors everything started working

RandomOakForest, 2025-03-07:
Thank you for following up. We were not able to reproduce this issue. If you find the issue persists with the latest Isaac Lab and Isaac Sim, please find a new _bug report_.

closed 2025-03-07 by RandomOakForest

philox12358, 2025-03-18:
The error: _AttributeError: 'Articulation' object has no attribute '_root_physx_view'._
I also encountered this error today, which was triggered when reading the properties of the robot. The solution is to ensure that only one member in your robot contains the **Articulation Root** attribute, and multiple or no **Articulation Root** attributes will result in this error.
If this solution solves the problem, please let me know.