isaac-sim/IsaacLab #1764: [Question] Disabling CUDA P2P Bandwitch and Latency Validation Test

itsmehere, 2025-01-31:
### Question

After the new Isaac Sim update, running `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/play.py --task Isaac-Reach-Franka-v0 --num_envs=1` leads to a bandwidth test

```
2025-01-31 09:50:05 [11,582ms] [Warning] [gpu.foundation.plugin] Detected IOMMU is enabled. Running CUDA peer-to-peer bandwidth and latency validation.
Unidirectional P2P=Enabled Bandwidth (P2P Writes) Matrix (GB/s)
   D\D     0      1      2      3      4      5      6      7 
     0 [...]
     ...
     7 [...]
P2P=Enabled Latency (P2P Writes) Matrix (us)
   GPU     0      1      2      3      4      5      6      7 
     0 [...]
     ...
     7 [...] 

   CPU     0      1      2      3      4      5      6      7 
     0 [...]
     ...
     7 [...] 
2025-01-31 09:55:00 [306,480ms] [Warning] [gpu.foundation.plugin] CUDA peer-to-peer observed bandwidth: 0.2 GB/s.
2025-01-31 09:55:00 [306,481ms] [Warning] [gpu.foundation.plugin] CUDA peer-to-peer observed latency: 49265.6 us.
2025-01-31 09:55:00 [306,481ms] [Warning] [gpu.foundation.plugin] Please verify if observed bandwidth and latency are expected
```

Overall, launching isaac sim's viz in non-headless mode seems to take a significantly longer period of time after the new update. Is there a way to disable this check - I don't think it was happening before. If not, is there an alternate recommended solution?

RandomOakForest, 2025-01-31:
Thank you for posting this. We will investigate.

itsmehere, 2025-02-01:
Thank you! My isaac sim and isaac lab versions are below:
```
$ conda list
isaacsim                  4.5.0.0                  pypi_0    pypi
...
isaaclab                  0.33.13                  pypi_0    pypi
```

closed 2025-03-07 by RandomOakForest