isaac-sim/IsaacLab #1773: [Question] Does the hdf5 data in the new IL example record the RGB image data?

Elycyx, 2025-02-01:
### Question

In the new example of imitation learning, does the hdf5 data record the RGB image data from the sensors? If not, how should the observation be handled when deploying the policy to a real robot?
Alternatively, if I want to add RGB information to the observation, how should I proceed?

RandomOakForest, 2025-02-07:
Thanks for posting this. The current implementation of the learning method is state-based and does not record the RGB data. We are reviewing deployment strategies you may pursue. Will aim to follow up soon.

Haiyuan-Liu, 2025-03-09:
Looking forward to the updated strategies. Thank you, @RandomOakForest.

RandomOakForest, 2025-03-19:
Following up, we will eventually have a tutorial that explains in more detail how to do this, but in the meantime, the team has provided [Isaac-Cartpole-RGB-v0](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab_tasks/isaaclab_tasks/manager_based/classic/cartpole/cartpole_camera_env_cfg.py) and [Isaac-Cartpole-RGB-Camera-Direct-v0](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab_tasks/isaaclab_tasks/direct/cartpole/cartpole_camera_env.py) as good examples on working with RGB data. Let us know if we can be of further help.

closed 2025-03-19 by RandomOakForest