isaac-sim/IsaacLab #1777: OmniHub only work for one linux user and needed to switch .cach path for another user to use(With Hot Fix Solution)

zoctipus, 2025-02-02:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Hurray to IsaacLab 2.0!

When the first user installed Isaac sim, either binary or pip install the  OmniHub can be launched correctly,

When the second user install Isaac sim, again, binary or pip, will not be able to launch  OmniHub when isaac-sim is launched,  and shows error message 

```
OmniHub: ThreadId(03) Hub encountered error. Trying to reconnect to Hub. retry_reason="Hub failed to launch: Io(\"child exited with exit status: 101 without writing file
```

It took me a while to find out the solution is to change the cache path in the account where the omniHub is correctly launched, clicking the CACHE:ON on the top right side of the IsaacSim GUI,  Then Change the Cache Path to correct user.

[image]

That is also saying, under one system, only one user can be able to use OmniHub, 
I realize this is issue of isaac sim not isaac lab, I want to post here and so people ran into same issue can find the tricky but easy solution!

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU: 4090
- CUDA: 12.5
- GPU Driver: 555

### Checklist

- [x ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-02-06:
Thank you for posting this. We will investigate.

dablro12, 2025-04-15:
Thank your effort for solving this problem. 👍

RandomOakForest, 2025-04-18:
Following up on this, it seems that the when running multiple instances of _Hub_, the cache directory should be different for each instance. You may be able to work with the env variables that Hub works with, specifically `OMNICLIENT_HUB_CACHE_DIR`. [See this doc for reference](https://docs.omniverse.nvidia.com/kit/docs/client_library/latest/index.html#hub).

closed 2025-07-07 by kellyguo11

Aiforall123, 2025-07-22:
Hello, how did you solve this problem?