isaac-sim/IsaacLab #1780: [Bug Report] Error when run isaac lab due to unable to spawn isaac sim

zakydzul, 2025-02-03:
## Bug Description

After following the official Local Installation in: [https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-sim](url), I'm unable to verify the Isaac Lab installation process, this is probably because it fails to loadthe isaac sim.
Here are some details:
1. When I run only the isaac sim only, it throws some error but the isaac sim still able to load.
2. Isaac lab can be executed in headless mode without any problem.
3. On another system that I have, I cannot reproduce this error.

## Steps to reproduce

1. Follow the Local Installation process in windows on Isaac Lab docs.
2. The error occurs in Verifying the Isaac Lab installation.

## Error Code 

Isaac sim error, but still able to load:
```
2025-02-03 06:46:00 [6,068ms] [Error] [carb.scripting-python.plugin] UnicodeDecodeError: 'cp950' codec can't decode byte 0xe2 in position 1094: illegal multibyte sequence

At:
  C:\Users\admin\anaconda3\envs\env_isaaclab\lib\json\__init__.py(293): load
  c:/users/admin/anaconda3/envs/env_isaaclab/lib/site-packages/isaacsim/exts/isaacsim.sensors.rtx.ui/isaacsim/sensors/rtx/ui/extension.py(54): on_startup
  C:\Users\admin\anaconda3\envs\env_isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(164): _startup_ext
  C:\Users\admin\anaconda3\envs\env_isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(224): startup
  C:\Users\admin\anaconda3\envs\env_isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  C:\Users\admin\anaconda3\envs\env_isaaclab\lib\site-packages\omni\kit_app.py(119): startup
  C:\Users\admin\anaconda3\envs\env_isaaclab\lib\site-packages\isaacsim\__init__.py(161): main
  C:\Users\admin\anaconda3\envs\env_isaaclab\Scripts\isaacsim.exe\__main__.py(7): <module>
  C:\Users\admin\anaconda3\envs\env_isaaclab\lib\runpy.py(86): _run_code
  C:\Users\admin\anaconda3\envs\env_isaaclab\lib\runpy.py(196): _run_module_as_main

2025-02-03 06:46:00 [6,068ms] [Error] [omni.ext.plugin] [ext: isaacsim.sensors.rtx.ui-0.1.2] Failed to startup python extension.
```
Isaac lab error code:
```
2025-02-03 06:50:44 [3,379ms] [Error] [carb.dictionary.serializer-toml.plugin] TOML parsing failed: Value must follow after a '=' at line 1
2025-02-03 06:50:44 [3,379ms] [Error] [omni.kit.registry.nucleus.utils.common] [kit/sdk] Failed pulling extension 'omni.kit.menu.edit-1.1.25+d02c707b'. Extension cache is corrupted: Failed to read: 'D:/omniverse/data/exts/v2/omni.kit.menu.edit-874360293737b28f\config/extension.gen.toml'. Cleaning the cache.
2025-02-03 06:50:44 [3,379ms] [Error] [carb.dictionary.serializer-toml.plugin] TOML parsing failed: Value must follow after a '=' at line 1
2025-02-03 06:50:44 [3,380ms] [Error] [omni.kit.registry.nucleus.utils.common] [kit/sdk] Failed pulling extension 'omni.kit.menu.edit-1.1.25+d02c707b'. Extension cache is corrupted: Failed to read: 'D:/omniverse/data/exts/v2/omni.kit.menu.edit-874360293737b28f\config/extension.gen.toml'. Cleaning the cache.
2025-02-03 06:50:44 [3,380ms] [Error] [omni.ext.plugin] Failed to pull extension: 'omni.kit.menu.edit-1.1.25' in 'c:/users/admin/anaconda3/envs/env_isaaclab/lib/site-packages/omni/data/kit/isaac-sim/4.5/exts/3'.
2025-02-03 06:50:44 [3,380ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "D:\IsaacLab\scripts\tutorials\00_sim\create_empty.py", line 29, in <module>
    app_launcher = AppLauncher(args_cli)
  File "D:\IsaacLab\source\isaaclab\isaaclab\app\app_launcher.py", line 114, in __init__
    self._create_app()
  File "D:\IsaacLab\source\isaaclab\isaaclab\app\app_launcher.py", line 638, in _create_app
    self._app = SimulationApp(self._sim_app_config, experience=self._sim_experience_file)
  File "C:\Users\admin\anaconda3\envs\env_isaaclab\Lib\site-packages\isaacsim\exts\isaacsim.simulation_app\isaacsim\simulation_app\simulation_app.py", line 227, in __init__
    from .utils import create_new_stage, open_stage
  File "C:\Users\admin\anaconda3\envs\env_isaaclab\Lib\site-packages\isaacsim\exts\isaacsim.simulation_app\isaacsim\simulation_app\utils.py", line 17, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'
```

## System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 4.5.0
- OS: Windows 10
- GPU: RTX 4080
- CUDA: 12.8
- GPU Driver: 572.16

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Entongsu, 2025-02-06:
I have encountered with similar problems

RandomOakForest, 2025-02-06:
Thank you for posting this. The team will look into it.

kellyguo11, 2025-02-06:
It may be because some of the extensions were corrupted or weren't able to be downloaded correctly. Might be worth retrying the pip installation process to let it re-download the extensions needed.

zakydzul, 2025-02-07:
I've tried to reinstall everything (anaconda env reinstall and git clone) several times, but unfortunately the problem still persists.

zakydzul, 2025-02-07:
> It may be because some of the extensions were corrupted or weren't able to be downloaded correctly. Might be worth retrying the pip installation process to let it re-download the extensions needed.

Yup, I tried to clear the cache in omniverse folders and it works!

closed 2025-02-07 by zakydzul