isaac-sim/IsaacLab #1781: [Question] Can someone please explain why there are scaled terms in the existing RL examples?

celestialdr4g0n, 2025-02-03:
### Question

I came across the Franka cabinet environment in Manager-based and Direct environment. and I saw that the dof position in the observatin of the Direct environment is scaled while the same term was not scaled in Manager-based enviroment

Afshin1997, 2025-02-05:
If I am not wrong, in manager-based environments the *empirical_normalization* parameter is set to **False**, whereas in direct environments it is set to **True**. Consequently, in direct environments it is better to scale the observation elements to a reasonable range (e.g., [-1, 1]). This can be achieved by multiplying each observation element by an appropriate scaling factor to ensure more uniform normalization. These scaling factors can be defined as:

$scale = \frac{1}{max(element)}$

celestialdr4g0n, 2025-02-07:
Thank you for your response, and I have no idea about empirical_normalization parameter. I heard of Reward Normalization for noise reduction, but I did not heard of State/Observation Normalization.

closed 2025-03-21 by RandomOakForest