isaac-sim/IsaacLab #1784: [Question] Sim2Real friction issues with Go2

Ashutosh781, 2025-02-03:
### Question

I am facing difficulties transferring policies from the simulator to the hardware for Unitree Go2. Primarily related to the actuator modeling and friction in the simulator. In the simulator, the policy learns to walk while dragging its feet or not lifting the feet high enough. Then, transferring this policy to hardware, the robot isn't able to move. Ideally, the policy should learn to lift the feet with the feet-air-time rewards and floor friction should not make it possible for the robot to drag the feet and still be able to move.

I think the issue is primarily due to the friction modeling of the floor in the simulator. Are there any ways to simulate high floor friction or randomize these values? I have even tried randomizing foot dynamic and static friction, but that doesn't help.

YannickP99, 2025-02-05:
Hey,
I have no  experience with sim2real yet, but it sounds like you used a plane-terrain. In case you have not heard of it, you can take a look at TerrainGenerator for example in the AnymalC locomotion demo. You can use this to generator bumps, stairs or slopes.
```
terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="generator",
        terrain_generator=ROUGH_TERRAINS_CFG,
        max_init_terrain_level=9,
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
        ),
        visual_material=sim_utils.MdlFileCfg(
            mdl_path="{NVIDIA_NUCLEUS_DIR}/Materials/Base/Architecture/Shingles_01.mdl",
            project_uvw=True,
        ),
        debug_vis=False,
    )
```

RandomOakForest, 2025-02-07:
Thanks for posting this. @YannickP99 has pointed you in the right direction. For reference, this is the [TerrainGenerator doc](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.terrains.html#isaaclab.terrains.TerrainGenerator).

Ashutosh781, 2025-02-07:
> Hey, I have no experience with sim2real yet, but it sounds like you used a plane-terrain. In case you have not heard of it, you can take a look at TerrainGenerator for example in the AnymalC [...]

Thanks for the answer; I was already aware of the rough terrain environments with steps and pyramid types of terrain. Since my current problem is limited to flat terrain with changing friction coefficients, I wanted a solution for the flat terrain environments. Introducing steps doesn't necessarily solve that problem.

Koo-bone-seok, 2025-02-07:
In velocity_env_cfg.py,  
class EventCfg:
    # startup
    physics_material = EventTerm(
        func=mdp.randomize_rigid_body_material,
        mode="startup",
        params={
            "asset_cfg": SceneEntityCfg("robot", body_names=".*"),
            "static_friction_range": (0.3, 1.25), #0.8 0.8
            "dynamic_friction_range": (0.3, 1.25), #0.6 0.6
            "restitution_range": (0.0, 0.0), # 0.0 0.0 # call when the object crash with other objects
            "num_buckets": 64,
        },
    )
 
this code is randomizing friction and resitution. you can change this for change ground friction coefficients. (static_friction_range and dynamic_friction_range)
and you can add "make_consistent" : True under the "num_buckets": 64, for ensure that dynamic friction does not exceed static friction.

I also deployed it on the Go2 robot, but I encountered an issue when the robot was walking. The robot keeps falling while walking, and it seems unstable If you manage to deploy it and confirm that it walks stably, could you give me some personal feedback later? I’d like to know if there are any differences between the methods you used for deployment and mine.

YannickP99, 2025-02-07:
I have read in some papers that using the plane-ground can lead to unnatural gaits. Using the terrain generator (not for stairs, but for uneven ground) forces the robot to actually lift its feet and not drag them (havent tested it yet because of some other bugs I received when using the TerrainGenerator)
@Ashutosh781

giulioturrisi, 2025-02-07:
I think there is some bug with the USD of go2. If I display the contact sensor during simulation, they are always red. I solved this by removing the cauliflower joint/body, and the robot has better behavior. Before, air time was practically zero.

YannickP99, 2025-02-07:
Hey @giulioturrisi, what joint or body did you remove? There might be a translation or typing error in your post

giulioturrisi, 2025-02-07:
calflower and calflower1, for each foot. Here the USD
https://drive.google.com/file/d/1tcOZLLPD2H9c9_RoDrL3kCCVzmTumZdW/view?usp=sharing

Ashutosh781, 2025-02-07:
Thanks a lot, everyone, for the replies; this discussion has been constructive.

- @Koo-bone-seok Yes, even I have used the physics material event to randomize friction values, although only for the feet' bodies instead of all of them. I trained with the ranges (0.8,2.0) for static and (0.6,1.6) for dynamic. This helped a lot where the robot lifted its feet and had some successful sim-to-real transfers. The gait wasn't perfect, and the robot was prone to falling. This might still need a bit more work to make it better.
- @YannickP99 Thanks for the feedback. I will give training with some minimal terrain a try to get a more natural gait.
- @giulioturrisi Thanks a lot for pointing out the error in the robot's USD file. I am currently trying to train policies with the file you provided to see if this helps with the sim-to-real aspects.

Again, thanks a lot, everyone, for the replies; I will post updates on what solutions helped.

giulioturrisi, 2025-02-07:
Please let me know if the new usd improves the sim-to-real as well!

YannickP99, 2025-02-08:
In case you manage to train a good policy with minimal terrain, I would love to hear about it @Ashutosh781

Mayankm96, 2025-02-15:
I am currently working on making sure sim-to-real works with IsaacLab for ANYmal-D robot. Will post my findings here once it is ready.

One of the main things I noticed, IsaacGym Previous 4.0 uses "average" combine mode for friction and restitition. If you look at our IsaacLab environments, we change the combine mode to be "multiply" because that's better to simulate smaller friction quantities. Thus, if you are porting parameters from IsaacGym to IsaacLab, this is something you should be aware of.

Why multiply is better? -- Suppose the foot is of friction 1.0 and ground is frictionless, i.e. 0.0. An average would mean the resulting friction is 0.5. However, we know this should be 0.0. Thus, we default to multiple in all the locomotion examples.

Besides that, I have noticed that the default command resampling is every 10.0 s, which isn't great. I would recommend reducing it to 4.0.

I hope these help! I'll post more once I get the policy to hardware.

Ashutosh781, 2025-02-27:
Thanks a lot, @Mayankm96, for the advice. I was able to achieve some initial sim-to-real transfers on Go2 using foot friction randomization. However, the policy was trained without any feet air time reward and achieved a pronking-like gait for the robot. 

I am having difficulty tuning the feet air time reward for Go2 to achieve a smoother trot gait. I have tried several values for the threshold and weight for the reward in a trial-and-error way. However, the policy does not seem to learn to take a step in simulation with the feet air time reward. Any suggestions for tuning the reward and having the robot lift its feet higher would be appreciated.

Mayankm96, 2025-03-01:
For Unitree robots, maybe check the reward weights in this repository: https://github.com/unitreerobotics/unitree_rl_gym

It appears they "tuned" it correctly enough to get sim-to-real working. If having the same reward weights doesn't work, we should check if there is an asset or some other issue.

Jdowdy05, 2025-03-13:
### Sim-to-Real Help and TL;DR
I am responding to this with a lot of what I've learned and how I've achieved some good sim-to-real results, as this question and its replies have people with a lot of the same/similar problems that I have had. I'm using Unitree's Go1 with a similar policy layout to the Spot example, since Go1 and Go2 are similar in size and weight, this policy should also work for all you Go2 users since the slight change in system dynamics shouldn't give different results during training.

In terms of getting the policy to lift its feet correctly, it's more or less all about reward shaping, which I will talk about below. However, having a wide range for both types of friction (0.3, 1.2), adding and setting "make_consistent" to true, as well as changing "restitution_range" to be (0.0, 0.15), helps with the sim-to-real transfer. If you use the spot example like I'm going to suggest, you'll want to change "reset_base" and "add_base_mass" to align with the robot's capabilities. My suggestions would be 1) use the spot example policy, it's a good starting point, and you can more easily modify rewards and their implementation. 2) Make or use an actuator-net made for your quadruped as this takes away a lot of the pain from sim-to-real. 3) lower the "sim.dt" to the point where decimation is around 10. i.e. set the physics dt to be 10 times faster than your policy dt. I'm not sure, but I'm guessing that increasing sim.dt helps the policy/network better capture some of the system dynamics that don't show up at a lower physics dt?

### We Have Liftoff
Here are some of the changes I made to the spot example that helped me achieve good results and stopped the feet from slipping/skidding on the ground. After playing around with the rewards and penalties for a while, the terms below seem important for getting quality locomotion.

| Term | Weight | Params |
| --- | --- | --- |
| air_time | 5.0 | mode_time: 0.15, velocity_threshold: 0.75 |
| foot_clearance | 1.5 |  target_height: 0.08 |
| action_smoothness | -1.25 | N/A |
| foot_slip | -1 | N/A |  

My intuition on this is that the increase in reward for "foot_clearance" and "air_time" seem to offset some of the energy minimization penalties from terms like "joint_torque", "joint_vel", and "joint_acc" while, the increase in penalty from "action_smoothness" and "foot_slip" tends to keep the policy from exploiting these reward increases.

### My Setup and Tips
I'm using a custom rough terrain along with the terrain curriculum; however, my terrain level drops to 0 near the beginning of training, which is nearly flat. The custom terrain I have removes stairs and tweaks parameters slightly for each sub-terrain. If you plan on using rough terrain, you'll need to implement your own "foot_clearance" reward since the pre-implemented version determines foot height relative to the world frame. I achieved this by determining the distance from the base to the foot in its initial and current poses in the z-axis as the foot and base frame are typically aligned. Another option is to use the height scanner sensor and the default "foot_clearance" reward implementation. As I said, my results visually seem good with my biggest issue being I don't have a good/accurate Kalman filter that I can run fast enough for its velocity state, making the quadruped sway slightly after moving. If any of you have a solution for that, lmk as I am in need😅 

P.S. If you'd like to see some non-optimal and rather funny ways of locomotion, you can keep all reward/penalty weights and parameters at their original values from the spot example and add a rough terrain curriculum. It's even better after transferred to the physical thing :)

RandomOakForest, 2025-03-21:
Thank you for following up and all the updates. I will close this issue for now but feel free to reopen it, or open a _discussion_ instead if we can be of further help. Thank you.

closed 2025-03-21 by RandomOakForest

YannickP99, 2025-03-27:
@Jdowdy05 thanks for your detailed description.  
I have one more question, how did you deploy the policy? I am struggling to find any information. Did you use the deployment code of the unitree_rl_gym repo?

Jdowdy05, 2025-03-27:
@YannickP99 I built my own script to run and deploy the policy, using onnx runtime and a Docker container to circumvent dependency issues. I had looked at Walk-These-Ways and go1-rl-kit as references, but neither exactly fit my needs. Depending on what you are trying to do you might be able to get away with using go1-rl-kit entirely. Hope this helps!