isaac-sim/IsaacLab #1794: [Question] Objects not spawning in the correct terrain environment

YannickP99, 2025-02-05:
### Question
I wanted to change from a plane terrain to a TerrainImporter terrain for better sim2real capabilities. In each of the environments I am spawning a robot and one RigidObject. Before everything worked fine but now I get the problem that the robots spawn correctly in each environment but the rigid objects are spawning in the global coordinate system (see screenshot).

[image]
I am using a DirectRLEnv, havent updated yet (isaacsim==4.2.0.2) because of some other bugs in the new version.
Code:
```
@configclass
class Go2EnvCfg(DirectRLEnvCfg):
    events: EventCfg = EventCfg()
    #* Simulation
    sim: SimulationCfg = SimulationCfg(
        dt=1 / 200,
        render_interval=decimation,
        disable_contact_processing=True,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
    )

    #* Terrain
    terrain_gen = TerrainGeneratorCfg(
                    size=(8.0, 8.0),
                    num_cols=16,
                    num_rows=1,
                    sub_terrains={
                        "random": HfRandomUniformTerrainCfg(
                        proportion=0.2, noise_range=(0.02, 0.05), noise_step=0.01
                        ),
    },
        
    )
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="generator",
        terrain_generator=terrain_gen,
        max_init_terrain_level=0,
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="multiply",
            restitution_combine_mode="multiply",
            static_friction=1.0,
            dynamic_friction=1.0,
        ),
        debug_vis=False,
    )

    #* Scene
    scene: InteractiveSceneCfg = InteractiveSceneCfg(num_envs=16, env_spacing=5.0, replicate_physics=False)

    #* Robot UNITREE_GO2_CFG
    # robot: ArticulationCfg = UNITREE_GO2_CFG.replace(prim_path="/World/envs/env_.*/Robot")     # type: ignore
    robot = ArticulationCfg(
        spawn=sim_utils.UrdfFileCfg(
            asset_path=f"source/thesis/resources/go2_manip_description/urdf/go2_manip_comb.urdf",    # go2_manip_reduced
            usd_dir=f"source/thesis/resources/go2_manip_description/usd",
            usd_file_name="go2_manip_exported",
            merge_fixed_joints=False,
            fix_base=False,
            force_usd_conversion=True,
            activate_contact_sensors=True,
            rigid_props=sim_utils.RigidBodyPropertiesCfg(
                disable_gravity=False,
                retain_accelerations=False,
                linear_damping=0.0,
                angular_damping=0.0,
[... 139 lines cut ...]
    def _setup_scene(self):
        #* Terrain
        self.cfg.terrain.num_envs = self.scene.cfg.num_envs
        self.cfg.terrain.env_spacing = self.scene.cfg.env_spacing
        self._terrain = self.cfg.terrain.class_type(self.cfg.terrain)
        #* Robot
        self._robot = Articulation(self.cfg.robot)
        self.scene.articulations["robot"] = self._robot
        #* Bins
        self._bin_0 = RigidObject(self.cfg.bin_0)
        self.scene.rigid_objects["bin_0"] = self._bin_0
        #* Sensors
        self._contact_sensor = ContactSensor(self.cfg.contact_sensor)
        self.scene.sensors["contact_sensor"] = self._contact_sensor
        #* Lights
        light_cfg = sim_utils.DomeLightCfg(intensity=2000.0, color=(0.75, 0.75, 0.75))
        light_cfg.func("/World/Light", light_cfg)
        #* Markers
        self._marker = VisualizationMarkers(self.cfg.markers_bin)
        # self._markers_bin_big = VisualizationMarkers(self.cfg.markers_bin_big)

        #* Clone, filter, and replicate
        self.scene.clone_environments(copy_from_source=True)
        self.scene.filter_collisions(global_prim_paths=[self._terrain.cfg.prim_path])
```

YannickP99, 2025-02-08:
@RandomOakForest a quick followup question. Is it possible to spawn one big rough environment like the standard plane where all robots spawn on the same plane instead of every robot spawning in an own environment like the method I tried before?

YannickP99, 2025-02-11:
@RandomOakForest I am sorry, the error was in my code!!

RandomOakForest, 2025-02-14:
Following up, there is a `_reset_idx` method you need to define, could you share this part of your code? To answer [your question](https://github.com/isaac-sim/IsaacLab/issues/1794#issuecomment-2644551889), the recommended way is to use environments as you are doing.

YannickP99, 2025-02-15:
Hey,
yes I found out about that. You mean similar to this, but for my objects?
```
      default_root_state = self._robot.data.default_root_state[env_ids]
      default_root_state[:, :3] += self._terrain.env_origins[env_ids]
      self._robot.write_root_link_pose_to_sim(default_root_state[:, :7], env_ids)
      self._robot.write_root_com_velocity_to_sim(default_root_state[:, 7:], env_ids)
      self._robot.write_joint_state_to_sim(
                              position=self._robot.data.default_joint_pos[env_ids],
                              velocity=self._robot.data.default_joint_vel[env_ids], 
                              joint_ids=None, env_ids=env_ids
                  )
```
Interestingly I do not need this for the simple plane-environment. After I implemented a similar addition like ```default_root_state[:, :3] += self._terrain.env_origins[env_ids]``` for my objects, I partially fixed my problem. So my objects are now spawning correctly, but I ran into the following thing:
I defined a terrain with 10 rows and 1 column and used 10 environments. So I thought that in each environment one robot would spawn. But some environments had multiple robots and some had none. Also in these environments the robot were spawning on the same position (also I defined a reset-event to randomize the robot position). So I wasnt sure how this could affect the training so I stopped and used the normal plane environment again
Thank you very much for your help!

StrainFlow, 2025-02-18:
It seems like you're trying to create a single procedural ground mesh and then spawn your 10 environments, each in the correct position on that mesh. to get a really good quadruped motion policy you're going to want to create more like 4,000 environments and this method probably won't scale very well.

that's why it's probably better to use the _reset_idx function. In this one, you would write code that generates each environment randomly so that you can generate as many random environments as you need. That said, I haven't done this myself just yet so I'm not sure if you can add additional meshes and rigid bodies in this function or if you can only write to the articulation's state. I will look into this

YannickP99, 2025-02-19:
@StrainFlow Thanks for your reply. I am aware of the number of environments. I have only been using 10 to visualize the environments and check if everything looks good, normally I am using 4096. When looking at the locomotion examples the ground meshes are generated in the EnvCfg-class and I think generating the ground mesh once is enough for my use case. Generating a new mesh on every reset sounds computationally expensive =)
I only do not understand how the robots are spawned in these environments. Maybe the robots are spawned in random environments intentionally.

Mayankm96, 2025-02-19:
Could you add a clone operation here to make sure the tensor is getting copied?

```python
      default_root_state = self._robot.data.default_root_state[env_ids].clone()
      default_root_state[:, :3] += self._terrain.env_origins[env_ids]
      self._robot.write_root_link_pose_to_sim(default_root_state[:, :7], env_ids)
      self._robot.write_root_com_velocity_to_sim(default_root_state[:, 7:], env_ids)
      self._robot.write_joint_state_to_sim(
                              position=self._robot.data.default_joint_pos[env_ids],
                              velocity=self._robot.data.default_joint_vel[env_ids], 
                              joint_ids=None, env_ids=env_ids
                  )
```

YannickP99, 2025-02-24:
I have tried out a few more things
1) In my code I am executing 
   ```
   default_root_state = self._robot.data.default_root_state[env_ids].clone()
   default_root_state[:, :3] += self._terrain.env_origins[env_ids]
   ``` 
   but I dont do the same operation for my RigidObject-class. For the plane environment everything looks fine.
   [image]

   When switching to the terrain the robots spawn as expected in the different environments (still some environments are empty but num robots = num environments) and the objects spawn around the global origin.
   [image]

2) When I add the same operation for the RigidObject-class, the environments looks different for the original plane environment. 
   [image]
   
   Also for the terrain it does not work correctly. Some objects are spawning with the defined env_spacing 
   [image]ng

StrainFlow, 2025-03-14:
Can you share your _reset_idx function?

YannickP99, 2025-03-15:
```
def _reset_idx(self, env_ids: torch.Tensor | None):
        if env_ids is None or len(env_ids) == self.num_envs:
            env_ids = self._robot._ALL_INDICES      # type: ignore

        self._robot.reset(env_ids)  
        super()._reset_idx(env_ids)                 # type: ignore

        if len(env_ids) == self.num_envs:                   # type: ignore
            # Spread out the resets to avoid spikes in training when many environments reset at a similar time
            self.episode_length_buf[:] = torch.randint_like(self.episode_length_buf, high=int(self.max_episode_length))
        self._actions[env_ids] = 0.0
        self._prev_actions[env_ids] = 0.0
        self._prev_prev_actions[env_ids] = 0.0     
        
        #* Reset robot state
        default_root_state = self._robot.data.default_root_state[env_ids]
        default_root_state[:, :3] += self._terrain.env_origins[env_ids]
        self._robot.write_root_link_pose_to_sim(default_root_state[:, :7], env_ids)
        self._robot.write_root_com_velocity_to_sim(default_root_state[:, 7:], env_ids)
        self._robot.write_joint_state_to_sim(
                                position=self._robot.data.default_joint_pos[env_ids],
                                velocity=self._robot.data.default_joint_vel[env_ids], 
                                joint_ids=None, env_ids=env_ids
```
That is the relevant part, after that I do some task reset and logging

RandomOakForest, 2025-04-02:
Thank you for following up. I will move this post to our _Discussions_ section for the team to continue. Let us know if you have made any inroads. Thanks.