isaac-sim/IsaacLab #1796: [Bug Report] Example Training Script Fails Due to Missing Assets and Contact Sensor Errors

zlw21gxy, 2025-02-05:
## Description
Following the [official pip installation guide](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-sim), I attempted to run the example training script, but it fails with asset loading and contact sensor errors.

Command used:
```bash
./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --headless
```

## Environment
- OS: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
- Kernel: 6.8.0-52-generic
- GPU: NVIDIA GeForce RTX 4090 (Driver: 535.183.01)
- CPU: 13th Gen Intel i7-13700KF
- RAM: 64GB
- Isaac Sim: 4.5

## Error Details

1. Asset Loading Error:
```
Warning: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/IsaacLab/Robots/ANYbotics/ANYmal-C/anymal_c.usd@
```

2. Contact Sensor Error:
```
ValueError: No contact sensors added to the prim: '/World/envs/env_0/Robot'. This means that no rigid bodies are present under this prim. Please check the prim path.
```

Notable warnings:
- `omni.isaac.dynamic_control` deprecated warning
- IOMMU enabled warning
- Failed to modify rigid body and articulation root properties

## Reproduction Steps
1. Followed the [pip installation guide](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-sim)
2. Cloned the IsaacLab repository
3. Ran the example training script
4. Error occurs during scene creation (~180 seconds into execution)

## Questions
1. Are there additional asset files that need to be downloaded? The ANYmal-C USD file seems to be missing.
2. Is there a missing configuration for contact sensors?
3. Is the `omni.isaac.dynamic_control` deprecation affecting functionality?

## Additional Notes
The simulation takes ~180 seconds before failing, suggesting possible initialization or asset loading issues. Both the robot asset configuration and contact sensor setup appear to be problematic.

RandomOakForest, 2025-02-06:
Thank you for posting this. We will investigate.

kellyguo11, 2025-02-06:
I suspect the asset loading error is the only problematic one. it seemed to have failed to find the asset from the server for some reason. One way you can try is to download the asset first from the S3 bucket path and modify the code in your local repo to point the asset file to the local downloaded copy. Or perhaps Hub (https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/asset_caching.html) may be a possible solution too.

Edify0991, 2025-02-25:
Could you tell how to "modify the code in your local repo to point the asset file to the local downloaded copy" in detail?I find a file ,source/isaaclab/isaaclab/utils/assets.py,but I don't know how to modify the path of carb.settings.get_settings().get("/persistent/isaac/asset_root/cloud")

CP-Nelson, 2025-03-03:
> Could you tell how to "modify the code in your local repo to point the asset file to the local downloaded copy" in detail?I find a file ,source/isaaclab/isaaclab/utils/assets.py,but I don't know how to modify the path of carb.settings.get_settings().get("/persistent/isaac/asset_root/cloud")

check the last reply https://github.com/isaac-sim/IsaacLab/issues/1814

closed 2025-07-07 by kellyguo11