isaac-sim/IsaacLab #1800: [Bug Report] URDF converter not working on model with STL colision (IsaacLab2.0, IsaacSim4.5)

linchangyi1, 2025-02-06:
### Describe the bug

The urdf_converter fails to properly process URDFs containing multiple links with STL file collisions.

When multiple links are connected via a **fixed** joint with **dont_collapse="true"**, and their collision models come from **STL** files, the urdf_converter still merges these links into a single link—even when the **"--merge-joints"** flag is used.

This issue occurs specifically in IsaacLab 2.0 with Isaac Sim 4.5. However, it works correctly in IsaacLab 1.3 with Isaac Sim 4.2.

### Steps to reproduce

#### The URDF used
I made a simple urdf where two boxes are connected with a fixed-type joint. In solidworks, it looks like:
[image]

I add the flag of **dont_collapse="true"** in the urdf:
```
<?xml version="1.0" encoding="utf-8"?>
<robot name="isaac_test">
  
  <link name="large_box">
    <inertial>
      <origin xyz="-7.4087E-19 -4.5954E-19 -4.3293E-19" rpy="0 0 0"/>
      <mass value="0.009"/>
      <inertia ixx="7.5E-07" ixy="-3.9031E-24" ixz="-1.1655E-22" iyy="1.35E-06" iyz="-1.7968E-39" izz="7.5E-07"/>
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry><mesh filename="package://isaac_test/meshes/large_box.STL"/></geometry>
      <material name=""><color rgba="0.79216 0.81961 0.93333 1"/></material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry><mesh filename="package://isaac_test/meshes/large_box.STL"/></geometry>
    </collision>
  </link>

  <joint name="fixed_joint" type="fixed" dont_collapse="true">
    <origin xyz="0 0 0.015" rpy="1.5708 0 0"/>
    <parent link="large_box"/>
    <child link="small_box"/>
    <axis xyz="0 0 0"/>
  </joint>

  <link name="small_box">
    <inertial>
      <origin xyz="-2.0808E-19 -1.7347E-18 1.7926E-18" rpy="0 0 0"/>
      <mass value="0.004"/>
      <inertia ixx="2.6667E-07" ixy="2.0216E-40" ixz="-1.4298E-23" iyy="1.6667E-07" iyz="0" izz="1.6667E-07"/>
    </inertial>
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry><mesh filename="package://isaac_test/meshes/small_box.STL"/></geometry>
      <material name=""><color rgba="0.79216 0.81961 0.93333 1"/></material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry><mesh filename="package://isaac_test/meshes/small_box.STL"/></geometry>
    </collision>
  </link>
</robot>
```

#### Test in IsaacLab2.0 and Isaac_Sim 4.5

Without **--merge-joints**, it looks fine:
```
python scripts/tools/convert_urdf.py ~/urdf_test/isaac_test/urdf/isaac_test.urdf ~/urdf_test/isaac_test/isaac_45_test_no_merge.usd
```
[image]
[image]

<br>

**With** **--merge-joints**, the converter still merges them into a single link even though **"dont_collapse"** is in the urdf file:

```
python scripts/tools/convert_urdf.py ~/urdf_test/isaac_test/urdf/isaac_test.urdf ~/urdf_test/isaac_test/isaac_45_test_merge.usd --merge-joints
```
[image]
[image]

<br><br>

#### Test in IsaacLab1.3 and Isaac_Sim 4.2
**With** **--merge-joints**, the converter processes them **separately** (which is desired):

```
python source/standalone/tools/convert_urdf.py ~/urdf_test/isaac_test/urdf/isaac_test.urdf ~/urdf_test/isaac_test/isaac_42_test_merge.usd --merge-joints
```
[image]
[image]

### Additional context
Apart from this bug, another bug relative to contact_sensor also occurs when using a USD generated from such URDFs in IsaacLab, as reported in https://github.com/isaac-sim/IsaacLab/issues/1760

I have uploaded the urdf used.
[urdf_test.zip](https://github.com/user-attachments/files/18683038/urdf_test.zip)

RandomOakForest, 2025-02-06:
Thank you for posting this. We are reviewing this discrepancy.

kellyguo11, 2025-02-10:
Thanks for reporting this. We have added a mergemeshcollision setting in the new URDF importer, which unfortunately is not exposed yet to APIs so we are not able to set it yet from the Isaac Lab side. We are working on adding this flag to the importer and configs so that this can be modified in user scripts.

GiulioRomualdi, 2025-04-02:
Hi @linchangyi1, I'm actually experiencing the same problem. What I noticed is that if I manually use the GUI to load the URDF the frames are exported correctly, considering the `dont_collapse="true"`. By any chance, have you found a possible solution?

linchangyi1, 2025-04-03:
Hi @GiulioRomualdi, I didn't fix it and finally used the old version of IsaacLab.

torydebra, 2025-05-05:
Any updates on this?

According to what I see now (Isaac lab 4.5, 2e6946afb9b26f6949d4b1fd0a00e9f4ef733fcc), I have a not updated version of the URDF importer in the GUI (2.3.10 vs 2.3.14). In the 2.3.10, I occur in the problem of merging the collision mesh as above. When updating to 2.3.14 from the GUI, this is solved but I occur in a new bug that puts an offset on all (of the entire imported robot) collision meshes, which result them to be translated of +20/30 cm in the z axis.
 
Moreover, the bug 1 is still present when using the script, even after updating from the GUI. Probably the GUI and the terminal script are based on different api?