isaac-sim/IsaacLab #1801: How Was humanoid_dance.npz Converted in the Updated AMP?

soungsill, 2025-02-06:
Hi,

I am reviewing the recently updated AMP section, but I couldn't find a script explaining how the following file was converted:

IsaacLab/source/isaaclab_tasks/isaaclab_tasks/direct/humanoid_amp/motions/humanoid_dance.npz

I remember seeing a similar file being used in IsaacGym before. In that case, the conversion process was provided in the code. However, the file used there seems to be different from the one in IsaacLab.

Could you share details on how this file was converted or what source file was used? I would like to apply a similar approach to another file.

I would really appreciate your response on this. Thanks!

RandomOakForest, 2025-02-14:
Thanks for posting this. To do this a custom script was used. We are reviewing its release. @Toni-SM for vis.

closed 2025-03-14 by RandomOakForest

StefanoBraghetto, 2025-10-28:
At this point I started to create my own implementation, but since I am more familiar with rl-games I believe I'll do it with that one.