isaac-sim/IsaacLab #1803: [Question]  After updating IsaacLab, the rendering is not displaying correctly.

JinMinChan, 2025-02-07:
After upgrading from IsaacLab 1.4.0 to 2.0.0, the rendering is not displaying correctly as shown in the image. Other example scripts render properly, but this issue only occurs with my script. Enabling path tracing fixes the rendering, but it significantly slows down performance.

[image]
[image]

right-chan, 2025-02-07:
I am experiencing the same issue. 
I am currently using Ubuntu 22.04 and have completed the installation following the instructions from this [link](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#).

The tutorials run without any issues. 
However, even in the default demo, when I enable the `--enable_cameras`option, I encounter the same camera issue as the original poster.

| command | screenshot |
| - | - |
| `python scripts/demos/arms.py` | [image] |
| `python scripts/demos/arms.py --enable_cameras` | [image] |

wzx16, 2025-02-07:
Same issue here

alanhuang1990, 2025-02-10:
This is a bug introduced by new isaac lab version in app configuration.  This is resolved by replace the rtx configuration of v2.0.0 by the old setting in v1.4.0:

replace the code in rectangle of 
[image] 
by the old settings in  

[image]

right-chan, 2025-02-10:
@alanhuang1990 Thank you! Your solution worked perfectly for me.

right-chan, 2025-02-10:
and I tested this on my setup using the `v2.0.0` of the `isaaclab.python.rendering.kit` file, and it worked well after making the following changes:

- **Line 57:** `rtx.directLighting.sampledLighting.denoisingTechnique = 0` 
→ **Commented out** (`# rtx.directLighting.sampledLighting.denoisingTechnique = 0`)
- **Line 62:** `rtx.domeLight.upperLowerStrategy = 4` 
→ **Changed to** `rtx.domeLight.upperLowerStrategy = 0`

alanhuang1990, 2025-02-10:
> and I tested this on my setup using the `v2.0.0` of the `isaaclab.python.rendering.kit` file, and it worked well after making the following changes:
>
> * **Line 57:** [...]

Your solution seems to be berrer :)   @right-chan

kellyguo11, 2025-02-11:
Thanks for reporting this! We did some render setting optimizations in the Isaac Lab 2.0 release that now approximate the dome lighting by default for better performance. These settings will however flatten the lighting and shadows in the renders. We are working on making some preset options for the future release to allow easier selection of quality vs performance render modes.

closed 2025-02-12 by RandomOakForest