isaac-sim/IsaacLab #1804: [Question] raycast sensor to measure objects generated in each environment.

Koo-bone-seok, 2025-02-07:
I was using this repository for reinforcement learning control of the Unitree A1.
I make some object for each environment like below(pictures) and add cube like this
CUBE_CFG = AssetBaseCfg(
    prim_path="/World/Origin.*/Cube",
    spawn=sim_utils.CuboidCfg(
        size=(3., 3., 0.5),
        # rigid_props=sim_utils.RigidBodyPropertiesCfg(),
        # mass_props=sim_utils.MassPropertiesCfg(mass=1.0),
        collision_props=sim_utils.CollisionPropertiesCfg(),
        visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(1.0, 1.0, 1.0)),
    ),
    init_state=AssetBaseCfg.InitialStateCfg(pos=(0, 0, 2)),
)
self.scene.obj = CUBE_CFG.replace(prim_path="{ENV_REGEX_NS}/OBJ")
[image]

i tried below

    height_scanner = RayCasterCfg(
        prim_path="{ENV_REGEX_NS}/Robot/base",
        offset=RayCasterCfg.OffsetCfg(pos=(0.0, 0.0, 20.0)),
        attach_yaw_only=True,
        pattern_cfg=patterns.GridPatternCfg(resolution=0.1, size=[1.6, 1.0]),
        debug_vis=True,
        'mesh_prim_paths=["{ENV_REGEX_NS}/OBJ/geometry/mesh"],'
    )
and have an error 'AttributeError: 'RayCaster' object has no attribute 'drift''.

or i add one asset (Cube) and tried 'mesh_prim_paths=["/World/Cube/"],' but i got the same error
i add cube like this.CUBE.func("/World/Cube", CUBE, translation=(0.0, 0.0, 1.0))
CUBE = sim_utils.CuboidCfg(
        size=(10., 10., 0.5),
        collision_props=sim_utils.CollisionPropertiesCfg(),
        visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(1.0, 1.0, 1.0)),
)

Is there a way to configure the raycast sensor to measure objects generated in each environment?
Or is there a way to add a single object and configure the raycast sensor to measure it?

Is there update for Raycast Sensor sensing multi static mesh or dynamic mesh?

edubya, 2025-02-08:
I am guessing that your objects do not all belong to the same mesh. I was getting similar errors until I merged all objects into a single mesh in Isaac Sim.

RandomOakForest, 2025-02-14:
Thanks for posting this. There have been no updates to the Raycast Sensor, it seems it as in version 1.4.0. [Tjhe current doc can be found here](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sensors.html#isaaclab.sensors.RayCaster).

AntoineRichard, 2026-08-27:
Thanks @Bonseok-Gu99 for the question, and sorry for the slow reply!

Isaac Lab now ships `MultiMeshRayCaster` / `MultiMeshRayCasterCfg`, whose `mesh_prim_paths` takes `RaycastTargetCfg` entries (or bare paths, including `{ENV_REGEX_NS}` targets) — that's the sensor for per-environment objects. It landed in https://github.com/isaac-sim/IsaacLab/pull/3298. One gotcha: for *moving* per-env objects, pass `RaycastTargetCfg` entries explicitly, since bare strings are converted with `track_mesh_transforms` disabled for backwards compatibility. The legacy `RayCaster` remains single-mesh. Closing as resolved — please reopen if it doesn't cover your setup.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard