isaac-sim/IsaacLab #1805: ModuleNotFoundError: No module named isaacsim.core

shr-eyas, 2025-02-07:
### ModuleNotFoundError: No module named `isaacsim.core`

```bash
(env_isaaclab) ubuntu@ip-172-31-0-229:~/IsaacLab$ python source/isaaclab_tasks/isaaclab_tasks/direct/franka_cabinet/franka_cabinet_env.py 
Traceback (most recent call last):
  File "/home/ubuntu/BCR/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/direct/franka_cabinet/franka_cabinet_env.py", line 10, in <module>
    from isaacsim.core.utils.stage import get_current_stage
ModuleNotFoundError: No module named 'isaacsim.core'
```
### Steps to reproduce

I had strictly followed the steps mentioned [here](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html
).

And the following commands to verify Isaac Sim and Lab installations are working.
```bash
isaacsim    
python scripts/tutorials/00_sim/create_empty.py
```

But when I am running this:
```bash
python source/isaaclab_tasks/isaaclab_tasks/direct/franka_cabinet/franka_cabinet_env.py
```

I am getting this error:
```bash
Traceback (most recent call last):
  File "/home/ubuntu/BCR/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/direct/franka_cabinet/franka_cabinet_env.py", line 10, in <module>
    from isaacsim.core.utils.stage import get_current_stage
ModuleNotFoundError: No module named 'isaacsim.core'
```

### System Info

- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU: TESLA T4
- CUDA Version: 12.8
- GPU Driver: 570.86.15

philox12358, 2025-02-10:
I have encountered the same problem.

philox12358, 2025-02-10:
I have encountered the same problem in code:

philox12358, 2025-02-10:
I have encountered the same problem in code:

`import isaacsim.core.utils.prims as prim_utils`

can not import

shr-eyas, 2025-02-10:
@philox12358 I speculate this is because the names are deprecated in version 4.5.0 but I am new to this so don't know how to fix that.

kellyguo11, 2025-02-11:
Hi there, you won't be able to run the environment scripts directly. Please run from one of the scripts in the `scripts/` folder, which are meant to run as standalone runner scripts.

closed 2025-02-11 by kellyguo11

amariaholson, 2025-04-05:
it seems that spawning objects has move to isaaclab.sim 

import isaaclab.sim as sim_utils

# spawn a red cone
    cfg_cone = sim_utils.ConeCfg(
        radius=0.15,
        height=0.5,
        visual_material=sim_utils.PreviewSurfaceCfg(diffuse_color=(1.0, 0.0, 0.0)),
    )

Now sure how to spawn a Xform object / Blank / Null object?