isaac-sim/IsaacLab #1814: [Bug Report] Bug title AttributeError: 'NoneType' object has no attribute 'GetPath'

ChuangYan2, 2025-02-08:
Hi, IsaacLab Team,

I got error when I run : ./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Flat-G1-v0 --headless

I can do it when I use isaaclab 1.4.0, not sure how to  fix the problem, can you help with that, thanks!

If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-02-13:
Thank you for posting this. Could you clarify the versions of the tools you are using. If you are using Isaac Lab 2.0, you will need Isaac Sim 4.5. Thanks.

FrostyForest, 2025-02-14:
same problem using Isaac Lab 2.0 + Isaac Sim 4.5

FrostyForest, 2025-02-15:
> same problem using Isaac Lab 2.0 + Isaac Sim 4.5

ok, i solve this problem using this method:[https://github.com/isaac-sim/IsaacLab/discussions/534#discussioncomment-9859173](url)

hojae-io, 2025-02-16:
@FrostyForest I get the same error. I cannot see the link, would you mind sharing your solution again?

FrostyForest, 2025-02-16:
> [@FrostyForest](https://github.com/FrostyForest) Hi, I get the same error. I cannot see the link you shared. Would you mind sharing your solution again?

First, download local assets following [https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_faq.html#isaac-sim-setup-assets-content-pack](url).
Second, go to IsaacLab/source/isaaclab/isaaclab/utils/assets.py then change the parameter of 'NUCLEUS_ASSET_ROOT_DIR =' to the path of your local asset.

zhichenghe, 2025-03-05:
I have meet the same issue  and I am using isaac lab 2.0  docker   https://isaac-sim.github.io/IsaacLab/main/source/deployment/docker.html     docker pull nvcr.io/nvidia/isaac-lab:2.0.1 .  And I am asloe change the parameter of 'NUCLEUS_ASSET_ROOT_DIR =' to the path of my local asset.     
but  I am still meet this issue.

[image]

[image]

[image]

FrostyForest, 2025-03-05:
> I have meet the same issue and I am using isaac lab 2.0 docker https://isaac-sim.github.io/IsaacLab/main/source/deployment/docker.html docker pull nvcr.io/nvidia/isaac-lab:2.0.1 . And I am asloe [...]

I think the path should be the folder which contains Isaac and NVIDIA.

[image]

Shenzhaolong1330, 2025-03-20:
> > [@FrostyForest](https://github.com/FrostyForest) Hi, I get the same error. I cannot see the link you shared. Would you mind sharing your solution again?
> 
> First, download local assets following https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_faq.html#isaac-sim-setup-assets-content-pack Second, go to IsaacLab/source/isaaclab/isaaclab/utils/assets.py then change the parameter of 'NUCLEUS_ASSET_ROOT_DIR =' to the path of your local asset.

lord! you saved my life, I have been trapped in this for hours, I decide to marry you!!!!

closed 2025-03-21 by RandomOakForest

zz7379, 2025-05-20:
Still not have all assets need for https://isaac-sim.github.io/IsaacLab/main/source/overview/teleop_imitation.html

2025-05-20 06:15:14 [34,260ms] [Warning] [isaaclab.envs.manager_based_env] The render interval (2) is smal
ler than the decimation (5). Multiple render calls will happen for each environment step. If this is not i
ntended, set the render interval to be equal to the decimation.                                           
2025-05-20 06:24:52 [612,140ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omn
iverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/envs/env_0/PackingTable/SM_CratePacking_Table_A1/Cra
te_02_GRP/sm_whitecorrugatedbox_b26_white_03>: Could not open asset @http://omniverse-content-production.s
3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/Props/PackingTable/props/sm_whitecorrugatedbox_b/sm_white
corrugatedbox_b26_white_01.usd@ for reference introduced by @http://omniverse-content-production.s3-us-wes
t-2.amazonaws.com/Assets/Isaac/4.5/Isaac/Props/PackingTable/packing_table.usd@</Root/SM_CratePacking_Table
_A1/Crate_02_GRP/sm_whitecorrugatedbox_b26_white_03>. (recomposing stage on stage @anon:0x5641e3a3a6a0:Wor
ld0.usd@ <0x5641e3a3f590>)