isaac-sim/IsaacLab #1815: [Bug Report] Bug title

psh9002, 2025-02-08:
### Describe the bug
https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/modify_direct_rl_env.html
From the example above, Isaac Lab environment tries to get configs from `site-packages/isaaclab/source/isaaclab_tasks/.....` not from `MY_ISAACLAB_DIR/source/isaaclab_tasks/.....`

I cannot run the RL codes at `scripts/reinforcement_learning` when I followed the instructions in the example, the learning file didn't run correctly.

### Steps to reproduce

- Commit: [e.g. bf28d97]
- Isaac Sim Version: 4.5
- OS: Ubuntu 20.04
- GPU: RTX 4090
- CUDA: 12.8
- GPU Driver: 570.86.10

RandomOakForest, 2025-02-13:
Thank you for posting this. We will review.

psh9002, 2025-02-17:
@RandomOakForest 
The problem was fixed when I reinstalled Isaac Lab ([link](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#installing-isaac-lab:~:text=./isaaclab.sh%20%2D%2Dinstall%20%23%20or%20%22./isaaclab.sh%20%2Di%22))

Also, my environment uses my cloned project directories after reinstall.