isaac-sim/IsaacLab #1816: [Bug Report] test_ray_caster_camera fails for some values of focal lengths

edubya, 2025-02-08:
### Describe the bug

Test `test_output_equal_to_usd_camera_when_intrinsics_set` fails when the intrinsic matrix and focal length are set to mimic an Intel Realsense camera (800 x 1280 with [focal length](https://www.intelrealsense.com/wp-content/uploads/2020/06/Intel-RealSense-D400-Series-Datasheet-June-2020.pdf) of 1.93 mm). 

I know that the focal length passed to the `.set_intrinsic_matrices` function should be in cm.

### Steps to reproduce
1. Modify the appropriate lines in `test_ray_caster_camera.py`. Use a small focal length.
2. Run the test: `python source/isaaclab/test/sensors/test_ray_caster_camera.py`

```
[snip]
test_output_equal_to_usd_camera_when_intrinsics_set (__main__.TestWarpCamera)
Test that the output of the ray caster camera is equal to the output of the usd camera when both are placed ... 2025-02-08 21:01:09 [18,277ms] [Warning] [isaaclab.sensors.ray_caster.ray_caster] The prim at path /World/Camera is not a physics prim! Using XFormPrim.
2025-02-08 21:01:09 [18,324ms] [Warning] [isaaclab.sensors.camera.camera] Camera aperture offsets are not supported by Omniverse. These parameters are ignored.
FAIL
[snip]
```

### System Info

Describe the characteristic of your environment:

- Commit: [01c8f25f90582acad6cc46c4d86597ebafadee49]
- Isaac Sim Version: 4.5.0-rc.36+release.19112.f59b3005.gl
- OS: Ubuntu 22.04
- GPU: RTX 3090
- CUDA: 12.4
- GPU Driver: 550.120

### Additional context

N/A

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-02-13:
Thank you for posting this. Could you share the full output log with warnings and errors? Thanks.

edubya, 2025-03-04:
```
[INFO][AppLauncher]: Loading experience file: /home/eric/Repos/IsaacLab/apps/isaaclab.python.headless.rendering.kit
[Warning] [simulation_app.simulation_app] fast shutdown not supported with jupyter notebooks
Loading user config located at: '/home/eric/.local/share/isaac-sim-standalone@4.5.0/kit/data/Kit/Isaac-Sim/4.5/user.config.json'

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.120       | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3090          | Yes: 0 |     | 24576   MB | 10de      | 0          |
|     |                                  |        |     |            | 2204      | 55c08f29.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-52-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: Intel(R) Core(TM) i7-14700K
| Bare Metal Cores: 20 | Bare Metal Logical Cores: 40
| Available Cores:  28 
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 47909 | Free Memory: 38845
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
test_camera_init (__main__.TestWarpCamera)
Test camera initialization. ... /home/eric/miniforge3/envs/isaac/lib/python3.10/asyncio/base_events.py:688: ResourceWarning: unclosed event loop <_UnixSelectorEventLoop running=False closed=False debug=False>
  _warn(f"unclosed event loop {self!r}", ResourceWarning, source=self)
ResourceWarning: Enable tracemalloc to get the object allocation traceback
ok
test_camera_init_intrinsic_matrix (__main__.TestWarpCamera)
Test camera initialization from intrinsic matrix. ... ok
test_camera_init_offset (__main__.TestWarpCamera)
Test camera initialization with offset using different conventions. ... ok
test_camera_resolution (__main__.TestWarpCamera)
Test camera resolution is correctly set. ... ok
test_camera_set_world_poses (__main__.TestWarpCamera)
Test camera function to set specific world pose. ... ok
test_camera_set_world_poses_from_view (__main__.TestWarpCamera)
Test camera function to set specific world pose from view. ... ok
test_depth_clipping (__main__.TestWarpCamera)
Test depth clipping. ... ok
test_intrinsic_matrix (__main__.TestWarpCamera)
Checks that the camera's set and retrieve methods work for intrinsic matrix. ... ok
test_multi_camera_init (__main__.TestWarpCamera)
Test multi-camera initialization. ... ok
test_output_equal_to_usd_camera_intrinsics (__main__.TestWarpCamera)
Test that the output of the ray caster camera and usd camera are the same when both are ... ok
test_output_equal_to_usd_camera_when_intrinsics_set (__main__.TestWarpCamera)
Test that the output of the ray caster camera is equal to the output of the usd camera when both are placed ... FAIL
test_output_equal_to_usdcamera (__main__.TestWarpCamera) ... Traceback (most recent call last):
  File "/home/eric/Repos/IsaacLab/source/isaaclab/isaaclab/sensors/sensor_base.py", line 66, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/eric/Repos/IsaacLab/source/isaaclab/isaaclab/sensors/sensor_base.py", line 273, in _initialize_callback
    self._initialize_impl()
  File "/home/eric/Repos/IsaacLab/source/isaaclab/isaaclab/sensors/camera/camera.py", line 486, in _initialize_impl
    self._update_intrinsic_matrices(self._ALL_INDICES)
  File "/home/eric/Repos/IsaacLab/source/isaaclab/isaaclab/sensors/camera/camera.py", line 580, in _update_intrinsic_matrices
    focal_length = sensor_prim.GetFocalLengthAttr().Get()
RuntimeError: Accessed schema on invalid prim
terminate called after throwing an instance of 'pybind11::error_already_set'
  what():  ReferenceError: weakly-referenced object no longer exists

At:
  /home/eric/.local/share/isaac-sim-standalone@4.5.0/exts/isaacsim.core.simulation_manager/isaacsim/core/simulation_manager/impl/simulation_manager.py(493): <lambda>
  /home/eric/.local/share/isaac-sim-standalone@4.5.0/extscache/omni.syntheticdata-0.6.10+d02c707b.lx64.r.cp310/omni/syntheticdata/scripts/SyntheticData.py(1454): _deactivate_node_rec
  /home/eric/.local/share/isaac-sim-standalone@4.5.0/extscache/omni.syntheticdata-0.6.10+d02c707b.lx64.r.cp310/omni/syntheticdata/scripts/SyntheticData.py(1689): deactivate_node_template
  /home/eric/.local/share/isaac-sim-standalone@4.5.0/extscache/omni.replicator.core-1.11.35+106.5.0.lx64.r.cp310/omni/replicator/core/scripts/annotators.py(816): detach
  /home/eric/Repos/IsaacLab/source/isaaclab/isaaclab/sensors/camera/camera.py(150): __del__
[... cut ...]
  /home/eric/miniforge3/envs/isaac/lib/python3.10/unittest/suite.py(84): __call__
  /home/eric/miniforge3/envs/isaac/lib/python3.10/unittest/runner.py(184): run
  /home/eric/miniforge3/envs/isaac/lib/python3.10/unittest/main.py(271): runTests
  /home/eric/miniforge3/envs/isaac/lib/python3.10/unittest/main.py(101): __init__
  /home/eric/Repos/IsaacLab/source/isaaclab/isaaclab/app/runners.py(19): run_tests
  /home/eric/Repos/IsaacLab/source/isaaclab/test/sensors/test_ray_caster_camera.py(938): <module>

Aborted (core dumped)

```

RandomOakForest, 2025-04-02:
Thank you for following up. The team will review this.

pascal-roth, 2025-04-03:
Thanks, @edubya, for pointing out this issue, I could reproduce it, and it is only happening when the focal length seems to be below 1.0 (tested for 0.193, 1.93, and 19.3, where the issue only occurs for 0.193) .  The reason why it seems to be coming from the IsaacSim site is that the focal length has no influence on the actual camera rays when the intrinsic matrix is given. While the warp camera stays constant, the USD camera results vary. 

I forward the issue to the IsaacSim Team.

kellyguo11, 2025-07-07:
I believe this issue should be addressed in Isaac Sim 5.0, please let us know if you are still facing issues.

AntoineRichard, 2026-08-27:
@edubya thanks for the precise repro with the focal-length sweep, and thanks @pascal-roth for chasing it down — the discrepancy came from the Isaac Sim USD camera at focal lengths below 1.0, not from Isaac Lab's ray caster.

On current `develop`, `test_output_equal_to_usd_camera_when_intrinsics_set` still parametrizes exactly the cases you reported — `(0.193, 0.20955)` and `(0.193, 20.955)` — with no xfail or skip, and it runs in the Isaac Sim CI job. Full disclosure though: the depth comparison is still gated behind a `FIXME` for `focal_length == 0.193` (`source/isaaclab/test/sensors/test_ray_caster_camera.py:979`) pending the USD/UsdGeom-side fix, so the case renders but isn't asserted yet.

Closing since the root cause sits upstream and we're tracking the assertion re-enable rather than the original report — please reopen if you'd rather keep it open until that lands.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard