isaac-sim/IsaacLab #1819: [Bug Report] URDF Importer bugs in v2.0.0

fan-ziqi, 2025-02-09:
### Describe the bug

URDF Importer bugs in v2.0.0:
1. After adding `dont_collapse="true"` to the fixed joint, it will still be merged
2. Unable to output one USD file, but instead output multiple USD files, which conflicts with [existing documents](https://isaac-sim.github.io/IsaacLab/main/source/how-to/import_new_asset.html#using-urdf-importer)

### System Info

- Commit: main
- Isaac Sim Version: v4.5.0
- OS: Ubuntu 20.04
- GPU: RTX 4090

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

GiulioRomualdi, 2025-04-02:
Hi all! I'm actually experienced 

> 1. After adding dont_collapse="true" to the fixed joint, it will still be merged

what I noticed is that importing the model from the GUI works

likesweetie, 2025-04-29:
how's going on with this issue? I'm also struggling to renoveting my old USD on new isaac lab 2.1