isaac-sim/IsaacLab #1823: [Bug Report] Isaac-Velocity-Flat-Spot-v0 with spot_with_arm.usd crashes

mschweig, 2025-02-09:
### Describe the bug

When starting a RSL-RL training with `--task=Isaac-Velocity-Flat-Spot-v0` and Spot with arm (spot_with_arm.usd) the simulation crashes when creating the scene. Getting PhysX plugin errors: Fetching GPU Narrowphase failed! 700

Same setup with `spot.usd` (without arm) works fine. 

### Steps to reproduce

https://github.com/mschweig/IsaacLab/blob/main/source/isaaclab_assets/isaaclab_assets/robots/spot.py

1. change SPOT_CFG usd_path to spot_with_arm
2. execute the Flat Velocity Task `./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Flat-Spot-v0
`
3. Isaac SIM loads and crashes with the following PhysX error after a few seconds. 

```
[INFO]: Scene manager:  <class InteractiveScene>
	Number of environments: 4096
	Environment spacing   : 2.5
	Source prim name      : /World/envs/env_0
	Global prim paths     : ['/World/ground']
	Replicate physics     : True
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
2025-02-09 08:50:13 [22,947ms] [Error] [omni.physx.plugin] PhysX error: Fetching GPU Narrowphase failed! 700
, FILE /builds/omniverse/physics/physx/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1697
2025-02-09 08:50:13 [22,947ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-02-09 08:50:13 [22,947ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuEventRecord failed with error 700
, FILE /builds/omniverse/physics/physx/source/gpucommon/include/PxgCudaUtils.h, LINE 40
2025-02-09 08:50:13 [22,947ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-02-09 08:50:13 [22,947ms] [Error] [omni.physx.plugin] PhysX error: SynchronizeStreams cuStreamWaitEvent failed with error 700
, FILE /builds/omniverse/physics/physx/source/gpucommon/include/PxgCudaUtils.h, LINE 46
2025-02-09 08:50:13 [22,947ms] [Error] [omni.physx.plugin] Cuda context manager error, simulation will be stopped and new cuda context manager will be created.
2025-02-09 08:50:13 [22,957ms] [Warning] [omni.physx.plugin] PhysX warning: /builds/omniverse/physics/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp, FILE /builds/omniverse/physics/physx/source/gpucommon/src/PxgCudaMemoryAllocator.cpp, LINE 51
^C^C^C^C^C^C^C^C^C^C^C^C./isaaclab.sh: line 274: 37429 Killed                  ${python_exe} "$@"

```

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 4.5
- Isaac Lab 2.0
- OS: Ubuntu 22.04
- GPU: RTX 4070
- CUDA: 12.4
- GPU Driver: 550.120

### Additional context

Ticket opened with same issue, but already closed:

https://github.com/isaac-sim/IsaacLab/issues/1403

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] RL training works with Spot Arm

mschweig, 2025-02-09:
Already tried increasing GPU buffers.
According to PhysX documentation it could be a collision issue: 

https://nvidia-omniverse.github.io/PhysX/physx/5.3.1/docs/RigidBodyCollision.html#broad-phase-collision-detection

RandomOakForest, 2025-02-13:
Thank you for posting this. The team will review.

dipampatel18, 2025-07-03:
I am facing the same issue. Were you able to find any fixes? @RandomOakForest @mschweig

RandomOakForest, 2025-07-03:
This is an old issue we thought we had resolved referring to an earlier version of the tools. Could you please post a new bug report with your current details? Thanks

closed 2025-07-07 by kellyguo11