isaac-sim/IsaacLab #1824: [Bug Report] Isaac Lab RSL RL framework doesn't work, others do.

MuammerBay, 2025-02-09:
### Describe the bug

When running any training scripts on the RSL RL Framework, such as the one from the installation guide
``` python scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0 --headless ```
it errors `ValueError: Reference at 'refs/heads/master' does not exist` (see complete errors below).
Other libraries such as RL_Games, SKRL and SB3 work just fine.

### Steps to reproduce

```
(isaaclab_2) C:\IsaacLab2\IsaacLab>python scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0 --headless
[INFO][AppLauncher]: Loading experience file: C:\IsaacLab2\IsaacLab\apps\isaaclab.python.headless.kit
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: 'c:/users/murme/miniconda3/envs/isaaclab_2/lib/site-packages/omni/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: c:/users/murme/miniconda3/envs/isaaclab_2/lib/site-packages/omni/logs/Kit/Isaac-Sim/4.5/kit_20250209_155810.log
2025-02-09 14:58:10 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-02-09 14:58:10 [405ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
2025-02-09 14:58:11 [985ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action 
is needed from end-users.

|---------------------------------------------------------------------------------------------|
| Driver Version: 537.58        | Graphics API: D3D12
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3090          | Yes: 0 |     | 24340   MB | 10de      | 0bde0000.. |
|     |                                  |        |     |            | 2204      | 0          |
|     |                                  |        |     |            | 2b        |            |
|=============================================================================================|
| OS: Windows 10 Home, Version: 10.0 (22H2), Build: 19045, Kernel: 10.0.19041.5369
| Processor: AMD Ryzen 5 5600X 6-Core Processor
| Cores: 6 | Logical Cores: 12
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 32694 | Free Memory: 15726
| Total Page/Swap (MB): 50090 | Free Page/Swap: 15773
|---------------------------------------------------------------------------------------------|
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.classic.ant.ant_env_cfg:AntEnvCfg
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.classic.ant.agents.rsl_rl_ppo_cfg:AntPPORunnerCfg
[INFO] Logging experiment in directory: C:\IsaacLab2\IsaacLab\logs\rsl_rl\ant
Exact experiment name requested from command line: 2025-02-09_15-58-16
Setting seed: 42
[INFO]: Base environment:
        Environment device    : cuda:0
        Environment seed      : 42
        Physics step-size     : 0.008333333333333333
        Rendering step-size   : 0.016666666666666666
        Environment step-size : 0.016666666666666666
[INFO]: Time taken for scene creation : 2.988643 seconds
[INFO]: Scene manager:  <class InteractiveScene>
        Number of environments: 4096
        Environment spacing   : 5.0
        Source prim name      : /World/envs/env_0
        Global prim paths     : ['/World/ground']
        Replicate physics     : True
[INFO]: Starting the simulation. This may take a few seconds. Please wait...
[INFO]: Time taken for simulation start : 1.622173 seconds
[INFO] Command Manager:  <CommandManager> contains 0 active terms.
+------------------------+
|  Active Command Terms  |
+--------+-------+-------+
| Index  | Name  |  Type |
+--------+-------+-------+
+--------+-------+-------+

[INFO] Recorder Manager:  <RecorderManager> contains 0 active terms.
+---------------------+
| Active Recorder Terms |
[... 147 lines cut ...]

Describe the characteristic of your environment:

- Commit: c4bec8fe0
- Isaac Sim Version: 4.5
- OS: Windows 10 Home
- GPU: RTX 3090
- CUDA: 11.8
- GPU Driver: 537.58

### Additional context

I have tried other libraries and bunch of examples, and only all the training scripts containing RSL RL don't work. This was also the case before the update on the 08.02.25 - "Switches to RSL-RL install from PyPI".
I have also tried to switch to master branch as the error indicated "ValueError: Reference at 'refs/heads/master' does not exist" and the same happened on CUDA 12.8 with driver version 552.86 before downgrading.

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Making RSL RL Framework work

RandomOakForest, 2025-02-13:
Thank you for posting this. The team will review.

Mayankm96, 2025-02-15:
Closing this to keep discussion in one place: #1840 and #1841

closed 2025-02-15 by Mayankm96

MuammerBay, 2025-02-22:
Hi @Mayankm96 and @RandomOakForest , I just want to say that my errors have been resolved. All frameworks (including rsl_rl) work just fine :)