isaac-sim/IsaacLab #1825: [Bug Report] Bug title

IlanElfen, 2025-02-10:
Hi,
I'm having trouble getting the Nvidia OpenXR integration to work.

Description:
When I try to use NVidia's OpenXR Experience by running the code [found here](https://github.com/isaac-sim/IsaacLab/blob/main/scripts/tools/record_demos.py), which loads another kit:
vars(args_cli)["experience"] = f'{os.environ["ISAACLAB_PATH"]}/apps/isaaclab.python.xr.openxr.kit', 
I get the "No module named 'omni.kit.usd'". 

I don't get this module error when passing keyboard input (it doesn't rely on OpenXR then and doesn't load that kit).

Setup:
- IsaacSim 4.2.0
- Ubuntu 22.04
- Nvidia GPU 3090
- CUDA 11.8
- GPU driver: 565.57.01

Steps to reproduce:
- Clone IsaacLab & Set up the environment
- In the main branch, run:
        ./isaaclab.sh -p scripts/tools/record_demos.py --task Isaac-Stack-Cube-Franka-IK-Rel-v0 --teleop_device handtracking --num_demos 10

Thank you!

closed 2025-02-11 by IlanElfen

makolon, 2025-03-03:
Hello!
How did you resolve this issue? If you don’t mind, could you share the solution?

KevinGeng3, 2025-05-14:
> I'm pretty sure the solution was making sure the extension was loaded in IsaacSim. If you search isaaacsim.xr in the extension search bar, an extension shows up. Install it. Let me know if that fixes it for you

Thank you for your advice, but it did not fix that bug for me. And I found that when I add --enable_cameras, this error would occur. But if I do not add --enable_cameras, everything is fine.

zhangjingze21, 2025-10-02:
> Thank you for your advice, but it did not fix that bug for me. And I found that when I add --enable_cameras, this error would occur. But if I do not add --enable_cameras, everything is fine.

Me too