isaac-sim/IsaacLab #1831: [Bug Report] Add sensor fail when Interactive scene spawning sequential

chlxydl, 2025-02-11:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug
1. On `interactive_scene.py` line 583, the code shows:
`for asset_name, asset_cfg in self.cfg.__dict__.items()`

the .items() means the item sequential is random, which means the spawn sequantial is random.

2. When adding a sensor to robot, it requires the robot being spawn first

3. If the sensor (such as Camera) being spawn before robot, if its prim_path is binded to the robot prim, it throwing "Unable to find source prim path /World/envs/env_.*/Robot/xxx, please create the prim before spawning."

**That's all, is there any interface to control the spawning sequential?**

### Steps to reproduce

### System Info

Describe the characteristic of your environment:

- Isaac Lab Version: 2.0.0
- Isaac Sim Version: 4.5

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

RandomOakForest, 2025-02-13:
Thank you for posting this. Since Python 3.7, the order of items returned by `items()` is guaranteed to be the insertion order. Could you post the full log of the output errors you are getting? Thanks.

chlxydl, 2025-02-16:
> Thank you for posting this. Since Python 3.7, the order of items returned by `items()` is guaranteed to be the insertion order. Could you post the full log of the output errors you are getting? Thanks.

I have checked it is ture that the order is inserction order. Maybe my problem becomes are there any way to control the insertion order?

For InteractiveSceneCfg, I can only define which variable I need, but no way to add a graph dependency information, was the order is the declare order? 

Thanks for your reply.

closed 2025-02-24 by chlxydl