isaac-sim/IsaacLab #1833: [Question] In the imitation learning Lift task, the target position during replay is inconsistent with the collected data.

Elycyx, 2025-02-11:
### Question

I replaced the task in the imitation learning example from `Isaac-Stack-Cube-Franka-IK-Rel-v0` to `Isaac-Lift-Cube-Franka-IK-Rel-v0` and made corresponding modifications to the Lift environment. Now, I can successfully collect human demonstration data through teleoperation. However, when replaying the collected dataset, I found that the target position did not match the data I collected. It seems that during replay, the target position is still being resampled and a new target is generated. ( I defined the target position using `CommandsCfg` in the same way as in the original Lift code.
```
@configclass
class CommandsCfg:
    """Command terms for the MDP."""

    object_pose = mdp.UniformPoseCommandCfg(
        asset_name="robot",
        body_name=MISSING,  # will be set by agent env cfg
        resampling_time_range=(5.0, 5.0),
        debug_vis=True,
        ranges=mdp.UniformPoseCommandCfg.Ranges(
            pos_x=(0.4, 0.6), pos_y=(-0.25, 0.25), pos_z=(0.25, 0.5), roll=(0.0, 0.0), pitch=(0.0, 0.0), yaw=(0.0, 0.0)
        ),
    )
```
)

Additionally, I checked the `initial_state` of the collected data and found the following content, which does not correctly include the target position.
```
initial_state {'articulation': {'robot': {'joint_position': tensor([[ 0.0000, -0.5690,  0.0000, -2.8100,  0.0000,  3.0370,  0.7410,  0.0400,
          0.0400]], device='cuda:0'), 'joint_velocity': tensor([[0., 0., 0., 0., 0., 0., 0., 0., 0.]], device='cuda:0'), 'root_pose': tensor([[0., 0., 0., 1., 0., 0., 0.]], device='cuda:0'), 'root_velocity': tensor([[0., 0., 0., 0., 0., 0.]], device='cuda:0')}}, 'rigid_object': {'object': {'root_pose': tensor([[ 0.4737, -0.1018,  0.0550,  1.0000,  0.0000,  0.0000,  0.0000]],
       device='cuda:0'), 'root_velocity': tensor([[0., 0., 0., 0., 0., 0.]], device='cuda:0')}}}
```
Could this be the cause of the issue? Or is there anything I can do to ensure that the target position remains consistent when using the collected data?

RandomOakForest, 2025-02-14:
Thanks for posting this. How are you doing the replay?

Elycyx, 2025-02-14:
> Thanks for posting this. How are you doing the replay?

I just follow the tutorial on the official [document](https://isaac-sim.github.io/IsaacLab/main/source/overview/teleop_imitation.html#imitation-learning).
```
# step a: replay the collected dataset
./isaaclab.sh -p scripts/tools/replay_demos.py --task Isaac-Stack-Cube-Franka-IK-Rel-v0 --dataset_file ./datasets/dataset.hdf5
```

Elycyx, 2025-02-18:
I think it might be that the command generated by `CommandsCfg` was not recorded when logging the data. I'm not sure if this is correct.

ayush055, 2025-02-20:
Hi, have you found any solution to this issue?

Elycyx, 2025-02-21:
> Hi, have you found any solution to this issue?

Not yet, I'm not very familiar with the Isaac Lab API, so I'm not sure how to proceed.

samuelvondi, 2025-03-07:
Did you try to train the BC agent with your collected data set using `./isaaclab.sh -p scripts/imitation_learning/robomimic/train.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0  --algo bc --dataset ./datasets/dataset_lift.hdf5`?

When doing so I get the following error:
```
run failed with error:
'Unable to synchronously open object (component not found)'

Traceback (most recent call last):
  File "/home/matthias/IsaacLab/scripts/imitation_learning/robomimic/train.py", line 326, in main
    train(config, device=device)
  File "/home/matthias/IsaacLab/scripts/imitation_learning/robomimic/train.py", line 159, in train
    trainset, validset = TrainUtils.load_data_for_training(config, obs_keys=shape_meta["all_obs_keys"])
  File "/home/matthias/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/robomimic/utils/train_utils.py", line 109, in load_data_for_training
    train_demo_keys = FileUtils.get_demos_for_filter_key(
  File "/home/matthias/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/robomimic/utils/file_utils.py", line 80, in get_demos_for_filter_key
    demo_keys = [elem.decode("utf-8") for elem in np.array(f["mask/{}".format(filter_key)][:])]
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "/home/matthias/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/h5py/_hl/group.py", line 360, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5o.pyx", line 257, in h5py.h5o.open
KeyError: 'Unable to synchronously open object (component not found)'
```
while everything works normally with the Stack example...

Elycyx, 2025-03-07:
> Did you try to train the BC agent with your collected data set using `./isaaclab.sh -p scripts/imitation_learning/robomimic/train.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0 --algo bc --dataset ./datasets/dataset_lift.hdf5`?
> 
> When doing so I get the following error:
> 
> ```
> run failed with error:
> 'Unable to synchronously open object (component not found)'
> 
> Traceback (most recent call last):
>   File "/home/matthias/IsaacLab/scripts/imitation_learning/robomimic/train.py", line 326, in main
>     train(config, device=device)
>   File "/home/matthias/IsaacLab/scripts/imitation_learning/robomimic/train.py", line 159, in train
>     trainset, validset = TrainUtils.load_data_for_training(config, obs_keys=shape_meta["all_obs_keys"])
>   File "/home/matthias/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/robomimic/utils/train_utils.py", line 109, in load_data_for_training
>     train_demo_keys = FileUtils.get_demos_for_filter_key(
>   File "/home/matthias/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/robomimic/utils/file_utils.py", line 80, in get_demos_for_filter_key
>     demo_keys = [elem.decode("utf-8") for elem in np.array(f["mask/{}".format(filter_key)][:])]
>   File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
>   File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
>   File "/home/matthias/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/h5py/_hl/group.py", line 360, in __getitem__
>     oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
>   File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
>   File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
>   File "h5py/h5o.pyx", line 257, in h5py.h5o.open
> KeyError: 'Unable to synchronously open object (component not found)'
> ```
> 
> while everything works normally with the Stack example... Does anyone know how this could be resolved?
Hi, It looks like there is an issue with your HDF5 file. When I encountered the previously mentioned problem, it was due to the target position being randomly generated while replaying the collected human demonstrations, causing all these episodes to be marked as failures. As a result, I could not use Isaac Lab Mimic to generate successful additional demonstrations. I am not sure if this is the reason for the issue with your HDF5 file.

samuelvondi, 2025-03-08:
I found it to be an issue with the config file of the BC algo from the Lift environment. You need to set all the keys to null for it to run.

RandomOakForest, 2025-04-02:
Thank you for following up. I will close this issue for now. If you find an inconsistency in our code base, please file a _bug report_. Thank you.

closed 2025-04-02 by RandomOakForest