isaac-sim/IsaacLab #1834: [Question] Question title

struggler176393, 2025-02-11:
Question

Hello there,

I'm trying to import Leap Hand URDF into USD with Isaac Sim URDF Importer. After deriving the USD file, I use ArticulationCfg class to instantiate the leap hand model, however, I found errors below:

`
RuntimeError: Failed to find an articulation when resolving '/World/envs/env_.*/Robot'. Please ensure that the prim has 'USD ArticulationRootAPI' applied.
`

Then, I set the leap hand prim as articulation root instead of the root joint, this error disappeared, while another error occurred:

`
2025-02-11 12:32:49 [15,706ms] [Warning] [omni.physx.tensors.plugin] Failed to find articulation at '/World/envs/env_0/Robot'
2025-02-11 12:32:49 [15,706ms] [Warning] [omni.physx.tensors.plugin] Failed to find articulation at '/World/envs/env_1/Robot'
2025-02-11 12:32:49 [15,706ms] [Error] [omni.physx.tensors.plugin] Pattern '/World/envs/env_*/Robot' did not match any rigid bodies
2025-02-11 12:32:49 [15,706ms] [Error] [omni.physx.tensors.plugin] Provided pattern list did not match any articulations
`

It seems that the USD file generated by the URDF Importer doesn't have articulation. I was hoping someone could lend me a hand. I'd be delighted to hear about it, thank you in advance!

Kind Regards,

Zhaohui Lin :)

KyleM73, 2025-02-11:
Usually when I see those errors it is because there is an issue either with the URDF or how I configured the importer. Are you able to share the URDF and the ArticulationCfg you used to generate the USD?

MissWu-615, 2025-02-14:
I have a similar questions when I try to run the command"./isaaclab.sh -p scripts/tutorials/03_envs/create_cartpole_base_env.py --num_envs 32",but there is an error "RuntimeError: Failed to find an articulation when resolving '/World/envs/env_.*/Robot'. Please ensure that the prim has 'USD ArticulationRootAPI' applied." Is it bacause there is some conflicts in my environment configuration or packages installed?

RandomOakForest, 2025-02-14:
Thanks for following up. It seems you may need a way to specify the primp path as in:
```
robot_cfg: ArticulationCfg = LEAP_HAND_CFG.replace(prim_path="/World/envs/env_.*/Robot")
```

itsmehere, 2025-02-20:
Same issue!

Error: `RuntimeError: Failed to find an articulation when resolving '/World/envs/env_.*/Robot'. Please ensure that the prim has 'USD ArticulationRootAPI' applied.`

ArticulationCfg:
```BIMANUAL_UR5_CFG = ArticulationCfg(
    spawn=sim_utils.UrdfFileCfg(
        asset_path=f"source/bimanual_lift/bimanual_lift/tasks/manipulation/lift/config/bimanual_ur5/ur5e_allegro/robots/dual_ur5e_allegro_real_v2.urdf",
        usd_dir=f"source/isaaclab_ext_assets/data/Robots/bimanual_ur5",
        usd_file_name=f"dual_ur5e_allegro_real_v2.usd",
        fix_base=True,
        visible=True,
        activate_contact_sensors=False,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            max_depenetration_velocity=5.0,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=True, solver_position_iteration_count=8, solver_velocity_iteration_count=0
        ),
        joint_drive=sim_utils.UrdfConverterCfg.JointDriveCfg(
            drive_type="force",
            target_type="position",
            gains=sim_utils.UrdfConverterCfg.JointDriveCfg.PDGainsCfg(stiffness=None, damping=None)
        ),
    ),

Junb0Dong, 2025-03-03:
Hi @struggler176393  I have same issue, I found this [solution](https://github.com/isaac-sim/IsaacLab/issues/959#issuecomment-2561969932), but I didn't find the option to add ArticulationRootAPI in isaacsim4.5 UI. Have you solve this problem?

StrainFlow, 2025-03-21:
> Hi [@struggler176393](https://github.com/struggler176393) I have same issue, I found this [solution](https://github.com/isaac-sim/IsaacLab/issues/959#issuecomment-2561969932), but I didn't find the option to add ArticulationRootAPI in isaacsim4.5 UI. Have you solve this problem?

Can you create your own thread and include a little more information about your specific case?

closed 2025-04-03 by RandomOakForest