isaac-sim/IsaacLab #1837: [Question] Migration to from isaaclab 1.4 to isaaclab 2.0

Takui9, 2025-02-11:
I am currently working on migration from isaaclab 1.4 to 2.0 and from isaac-sim 4.2 to 4.5. I am encountering an issue when deploying the old RL policy trained before on isaac-sim 4.5. 

The policy is for a very simple manipulation task, which is to lift the cube placed on the table to the target pose. I used the similar training setup provided in isaaclab_tasks/manager_based/manipulation/lift. The observation space included the robot joint velocity / position, object pose, target pose and the action, and I only replaced the robot model with my own. 

However, my old policy completely fails to work and outputs nonsense joint actions in the new simulation environment. I am just curious if the update on the isaac-sim can really lead to such a huge sim2sim gap, or I just missed modifying something during the merging?

kellyguo11, 2025-02-12:
Thanks for reporting this! In general, I don't think this is something we would expect. Would you be able to test with the Isaac Lab v1.4.1 (https://github.com/isaac-sim/IsaacLab/releases/tag/v1.4.1) release and see if your policy works there? That can help us narrow down on some recent changes that went into Isaac Lab.

closed 2025-03-02 by Mayankm96

Mayankm96, 2025-03-06:
We should have fixed this in 2.0.2. let us know if the problem persists for you!

PR #1873 (2025-03-02): Reverts the setting of joint velocity limits for implicit actuators
# Description

The previous fix in #1654 was checking if `effort_limits_sim` is None instead of checking `cfg.effort_limits_sim` for explicit actuators. This did NOT work as effort limits sim is a tensor that gets assigned the value on initialization.

The new fix now adds an implicit/explicit model attribute to the actuator model to ensure the right defaults are getting set. It moves all the implicit actuator warning code to its class to keep the articulation class cleaner.

We also revert the behavior of setting velocity lim [... cut]
commit 3d836ab1 (2025-03-02): Reverts the setting of joint velocity limits for implicit actuators (#1873)

# Description

The previous fix in #1654 was checking if `effort_limits_sim` is None
instead of checking `cfg.effort_limits_sim` for explicit actuators. This
did NOT work as effort limits sim is a tensor that gets assigned the
value on initialization.

The new fix now adds an implicit/explicit model attribute to the
actuator model to ensure the right defaults are getting set. It moves
all the implicit actuator warning code to its class to keep the
articulation class cleaner.

We also revert the behavior of setting vel [... cut]