isaac-sim/IsaacLab #1844: [Bug Report] pixelated flickering for hospital world with 2.0 release

babakc, 2025-02-12:
I'm using the hospital world for my isaaclab simulation and had no issues in IS4.2/IL1.4, however since upgrading to IS4.5/IL2.0, the hospital world has a strange flickering issue as I move the camera around. It flickers for about 10 seconds then settles till i move the camera around again. It doesn't seem to be a problem with the other environments, and is not a problem in just IS4.5. It's only when i load the world into isaaclab, that i get the flickering.

To reproduce the issue:
./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
Drag and drop the hospital.usd from the asset browser, then navigate the world. The stairs and the roof in particular flicker.

Screewn rec of the issue in isaaclab 2.0 is [here](https://drive.google.com/file/d/1uZcPQw_dx5HGJMYWr2DqUT5tb3cdapzv/view?usp=sharing)
For comparison, what it looks like in isaacsim 4.5 [here](https://drive.google.com/file/d/1ooCry9eLduqkWk2CjX11Qc1es6PyvgF3/view?usp=sharing) 

- Commit: 157a19e
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: nvidia L4 (AWS G6 instance)
- CUDA: 12.2
- GPU Driver: 535.216.03

RandomOakForest, 2025-02-13:
Thanks for posting this. Would it be possible for you to try the recommended driver `535.129.03`? Thanks.

babakc, 2025-02-14:
Unfortunetely that driver version isn't supported by my kernel `6.8.0-1021-aws`, similar to this issue [here](https://forums.developer.nvidia.com/t/nvidia-driver-nvidia-linux-x86-64-535-129-03-run-installation-failed-on-ubuntu-22-04/304564)

I see you've linked a pull request, is that the fix? I'll try this when i get back to my machine tomorrow if so.

kellyguo11, 2025-02-14:
Do you also see the same behavior when loading the USD scene in Isaac Sim?

babakc, 2025-02-15:
No, the rendering is normal in Isaac Sim. The issue only occurs in the latest Isaac Lab release.

PR #1855 (2025-03-05): Reverts to old render settings in kit experience files
# Description

Reverts to old render settings.

Fixes #1844 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

Before:
[image]

After:
[image]

## Checklist

- [ ] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [ ] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that my feature works
- [ ] I have updated the changelog and the corresponding version  [... cut]
commit 8b7fc5f8 (2025-03-05): Reverts to old render settings in kit experience files (#1855)

# Description

Reverts to old render settings.

Fixes #1844 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Screenshots

Before:

[image]

After:

[image]

## Checklist

- [ ] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [ ] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that my
feature works
- [ ] I have updated the changelog and the correspondi [... cut]
closed 2025-03-05 by kellyguo11