isaac-sim/IsaacLab #1849: [Question] How to run train.py on a specific GPU?

KevinGeng3, 2025-02-13:
### Question

When running training code using IsaacLab, I'm having trouble specifying which GPU to use for training. I have access to a server with 8*4090, here is the output of nvidia-smi:
```
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.216.03             Driver Version: 535.216.03   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4090        Off | 00000000:4F:00.0 Off |                  Off |
| 30%   32C    P2              65W / 450W |  24188MiB / 24564MiB |      7%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce RTX 4090        Off | 00000000:52:00.0 Off |                  Off |
| 30%   33C    P2              54W / 450W |   3336MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   2  NVIDIA GeForce RTX 4090        Off | 00000000:56:00.0 Off |                  Off |
| 30%   27C    P8              16W / 450W |   3066MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   3  NVIDIA GeForce RTX 4090        Off | 00000000:57:00.0 Off |                  Off |
| 30%   28C    P8              20W / 450W |   9582MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   4  NVIDIA GeForce RTX 4090        Off | 00000000:CE:00.0 Off |                  Off |
| 30%   27C    P8              10W / 450W |   3066MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   5  NVIDIA GeForce RTX 4090        Off | 00000000:D1:00.0 Off |                  Off |
| 30%   28C    P8              15W / 450W |   3066MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   6  NVIDIA GeForce RTX 4090        Off | 00000000:D5:00.0 Off |                  Off |
| 30%   29C    P8              16W / 450W |   3066MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   7  NVIDIA GeForce RTX 4090        Off | 00000000:D6:00.0 Off |                  Off |
| 30%   34C    P2              52W / 450W |   4228MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
```
And I want to use IPPO of SKRL and choose cuda:1 to run training code on. So I add one line in /workspace/isaaclab/scripts/reinforcement_learning/skrl/train.py:
```
def main(env_cfg: ManagerBasedRLEnvCfg | DirectRLEnvCfg | DirectMARLEnvCfg, agent_cfg: dict):
    """Train with skrl agent."""
    # override configurations with non-hydra CLI arguments
    env_cfg.scene.num_envs = args_cli.num_envs if args_cli.num_envs is not None else env_cfg.scene.num_envs
    env_cfg.sim.device = args_cli.device if args_cli.device is not None else env_cfg.sim.device
    env_cfg.sim.device = "cuda:1" # <----------------   CODE THAT I ADDED 
    # multi-gpu training config
    if args_cli.distributed:
        env_cfg.sim.device = f"cuda:{app_launcher.local_rank}"
    # max iterations for training
    if args_cli.max_iterations:
        agent_cfg["trainer"]["timesteps"] = args_cli.max_iterations * agent_cfg["agent"]["rollouts"]
    agent_cfg["trainer"]["close_environment_at_exit"] = False
    # configure the ML framework into the global skrl variable
    if args_cli.ml_framework.startswith("jax"):
        skrl.config.jax.backend = "jax" if args_cli.ml_framework == "jax" else "numpy"
```
Here is the full output:
```[INFO] Using python from: /workspace/isaaclab/_isaac_sim/python.sh                                                                                                                                                      
[INFO][AppLauncher]: Loading experience file: /workspace/isaaclab/apps/isaaclab.python.headless.kit
Loading user config located at: '/isaac-sim/kit/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/4.5/kit_20250213_120113.log
2025-02-13 12:01:13 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-02-13 12:01:14 [1,128ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
[... 157 lines cut ...]
    from . import mdp, ui
  File "/workspace/isaaclab/source/isaaclab/isaaclab/envs/mdp/__init__.py", line 18, in <module>
    from .actions import *  # noqa: F401, F403
  File "/workspace/isaaclab/source/isaaclab/isaaclab/envs/mdp/actions/__init__.py", line 8, in <module>
    from .actions_cfg import *
  File "/workspace/isaaclab/source/isaaclab/isaaclab/envs/mdp/actions/actions_cfg.py", line 9, in <module>
    from isaaclab.managers.action_manager import ActionTerm, ActionTermCfg
  File "/workspace/isaaclab/source/isaaclab/isaaclab/managers/__init__.py", line 13, in <module>
    from .action_manager import ActionManager, ActionTerm
  File "/workspace/isaaclab/source/isaaclab/isaaclab/managers/action_manager.py", line 20, in <module>
    from isaaclab.assets import AssetBase
  File "/workspace/isaaclab/source/isaaclab/isaaclab/assets/__init__.py", line 41, in <module>
    from .articulation import Articulation, ArticulationCfg, ArticulationData
  File "/workspace/isaaclab/source/isaaclab/isaaclab/assets/articulation/__init__.py", line 8, in <module>
    from .articulation import Articulation
  File "/workspace/isaaclab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 21, in <module>
    import isaaclab.sim as sim_utils
  File "/workspace/isaaclab/source/isaaclab/isaaclab/sim/__init__.py", line 32, in <module>
    from .simulation_context import SimulationContext, build_simulation_context  # noqa: F401, F403
ModuleNotFoundError: No module named 'isaaclab.sim.simulation_context'
2025-02-13 12:01:24 [11,057ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-02-13 12:01:24 [11,585ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```
I am sure 24GB GPU memory is enough for this task and the code is correct, because I have a 3090 PC, and I run the training code on it successfully, which only required about 10GB GPU memory.
Anybody knows the solution? Thank you in advance!

KevinGeng3, 2025-02-13:
Sorry, forgot to say my version of Isaaclab and Isaacsim: 2.0&4.5

kellyguo11, 2025-02-13:
Try setting CUDA_VISIBLE_DEVICES to the device you'd like to use.

KevinGeng3, 2025-02-13:
> Try setting CUDA_VISIBLE_DEVICES to the device you'd like to use.

Thank you! But I have tried 2 ways. First I add CUDA_VISIBLE_DEVICES=1 after "./isaaclab.sh -p scripts/reinforcement_learning/skrl/train.py --task Isaac-Dual-Arm-Reach-Direct-v1 --headless", it does not change the output. Then I add "os.environ['CUDA_VISIBLE_DEVICES'] = '1' " in scripts/reinforcement_learning/skrl/train.py, it does not change the output as well.
And I notice that there are errors: "ModuleNotFoundError: No module named 'isaaclab.sim.simulation_context'"  and "[Error] [omni.physx.plugin] Failed to get primary context of requested CUDA device 0" in the output. As you can see, cuda:0 is running out of its memory. So does cuda:0 running out of memory cause this problem?

kellyguo11, 2025-02-13:
It's probably best to set CUDA_VISIBLE_DEVICES before launching the training script. This will limit the access of GPUs for Isaac Sim to see just the device that was set through CUDA_VISIBLE_DEVICES, which should help get it off of device 0.